square/pylink #10: OSError: [WinError 193] %1 is not a valid Win32 application

rils, 2017-11-21:
I Installed Jlink seggar and i am able to connect over SWD using Jlink commander tool. So Jlink installation works fine. I installed python 3.6, pylink-square. I am able to import pylink. But getting `OSError` while crating JLink object. Please help.
`C:\Users\nxf41542>python
Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 18:11:49) [MSC v.1900 64 bit (AMD64)]
 on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>>
>>> import pylink
>>> pylink.JLink()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python36\lib\site-packages\pylink_square-0.0.7-py3.6.egg\pylink\jlink
.py", line 249, in __init__
  File "C:\Python36\lib\site-packages\pylink_square-0.0.7-py3.6.egg\pylink\libra
ry.py", line 239, in __init__
  File "C:\Python36\lib\site-packages\pylink_square-0.0.7-py3.6.egg\pylink\libra
ry.py", line 277, in load_default
  File "C:\Python36\lib\site-packages\pylink_square-0.0.7-py3.6.egg\pylink\libra
ry.py", line 331, in load
  File "C:\Python36\lib\ctypes\__init__.py", line 426, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python36\lib\ctypes\__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application`

rils, 2017-11-21:
I understood the problem. My PC is running on 64bit Win7 OS. SO I selected 64bit python version 3.6. But the JlinkARM.Dll is for 32bit. So python could not load it. I uninstalled python and installed 32bit version. Now Everything works good.! Thanks.

closed 2017-11-21 by rils

saeugetier, 2018-01-04:
Solution would be that on 64bit windows systems not JLinkARM.dll should be used. Please use JLink_x64.dll

The code could look something like that: 
`WINDOWS32_JLINK_SDK_NAME = 'JLinkARM'
WINDOWS64_JLINK_SDK_NAME = 'JLink_x64'

    if platform.architecture()[0]=='32bit':
            WINDOWS_JLINK_SDK_NAME = WINDOWS32_JLINK_SDK_NAME
    elif platform.architecture()[0]=='64bit':
         WINDOWS_JLINK_SDK_NAME = WINDOWS64_JLINK_SDK_NAME`

zamora18, 2018-02-16:
@saeugetier what I did was alter the file in `pylink/library.py` and changed line 87 to:
```python
WINDOWS_JLINK_SDK_NAME = 'JLink_x64'
```
Is this just as acceptable? I don't want it to cause problems later on.

hkpeprah, 2018-02-16:
Ah, interesting.  This was an oversight on our part.  If you want to submit a patch for this, that would be great!

saeugetier, 2018-02-16:
@zamora18 yes, you can do this. But then the code will only run on a 64bit system.

zamora18, 2018-02-16:
I think @saeugetier's addition to select between 32-bit and 64-bit systems would be a good solution.

I am not sure if this error is related to my change, but I get this output:
```
No handlers could be found for logger "pylink.jlink"
Traceback (most recent call last):
  File "C:/Users/zamora/PycharmProjects/jflash/my_flash_test.py", line 20, in <module>
    jlink.connect("STM32F446RE", verbose=True)
  File "C:\Users\zamora\PycharmProjects\jflash\pylink\jlink.py", line 141, in wrapper
    return func(self, *args, **kwargs)
  File "C:\Users\zamora\PycharmProjects\jflash\pylink\jlink.py", line 912, in connect
    raise errors.JLinkException(result)
pylink.errors.JLinkException: Unspecified error.
```
I have been trying to trace the error, because `Unspecified error` is not very clear.
The last portion of the error occurs after this call in `jlink.py`:
```python
result = self._dll.JLINKARM_Connect()
        if result < 0:
            raise errors.JLinkException(result)
```
I can't tell if it is actually looking for `JLINKARM` still since I changed to `JLINK_x64`? Could you help me with this? Thank you.

hkpeprah, 2018-02-16:
There's some documentation here around this error: http://pylink.readthedocs.io/en/latest/troubleshooting.html

It's a generic catch-all.

zamora18, 2018-02-16:
Thank you, I looked over those errors, and in my case I am attempting to __connect__ to the target when it fails. I believe I may be _"trying to connect to the target over JTAG when it only supports SWD"_. 

However, [in the documentation](http://pylink.readthedocs.io/en/latest/swd.html?highlight=swd), it states that I need to be connected to the target __before__ I can call `swo_start()`. I double-checked the other possible diagnoses for my error and it does not seem to be any of them. 
It has been difficult to diagnose because I have used the J-Link JFlash software and it connects & programs just fine (see image). One more detail to note, is that I __have__ gotten my python script to work, but it is very seldom and sporadic. What I usually try to do is use the J-Flash utility and erase the chip multiple times, and power cycle the device over and over.

I really appreciate any help. 

[image]

hkpeprah, 2018-02-17:
`swo_start` is for using Serial-Wire Output (SWO), you don't want to use it here (unless you want to do SWO stuff as well).  You want to connect over SWD:

```
jlink = pylink.JLink()
jlink.open()
jlink.set_tif(JLinkInterfaces.SWD)
jlink.connect('STM32F446RE')
```

zamora18, 2018-02-20:
Using `jlink.set_tif(pylink.JLinkInterfaces.SWD)` worked! 

Thank you for your help!

zamora18, 2018-08-08:
I am setting up a new machine (still getting this same error on 64-bit), and can't seem to modify the `pylink/library.py` file. It gives me a _File is Read Only_ message. @saeugetier  Where can I access this file to modify? 
I installed pylink into python by downloading the ZIP and running `python setup.py install`

Thanks

zamora18, 2018-08-08:
@hkpeprah [This PR](https://github.com/square/pylink/pull/25) **does** fix my issue and I no longer get the `OSError: [WinError 193] %1 is not a valid Win32 application` error message.

TiagoSousa26, 2019-07-08:
I am still getting this error. It seems to not be fixed :(

jjaredrides, 2020-05-19:
File "<string>", line 1, in <module>

  File "c:\users\yv999fy\onedrive - ey\documents\aflac_wwhq\model results\output_total\Compare.py", line 370, in compare_care

    import xlwings as xw

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\xlwings\__init__.py", line 9, in <module>

    from . import _xlwindows as xlplatform

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\xlwings\_xlwindows.py", line 25, in <module>

    from comtypes import IUnknown

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\comtypes\__init__.py", line 1176, in <module>

    class IPersist(IUnknown):

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\comtypes\__init__.py", line 1181, in IPersist

    ( ['out'], POINTER(GUID), 'pClassID' )),

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\comtypes\__init__.py", line 1099, in COMMETHOD

    from comtypes.automation import VARIANT

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\comtypes\automation.py", line 12, in <module>

    from comtypes import npsupport

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\comtypes\npsupport.py", line 5, in <module>

    import numpy

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\numpy\__init__.py", line 140, in <module>

    from . import _distributor_init

  File "C:\Users\YV999FY\AppData\Roaming\Python\Python37\site-packages\numpy\_distributor_init.py", line 26, in <module>

    WinDLL(os.path.abspath(filename))

  File "C:\Users\YV999FY\Anaconda3\lib\ctypes\__init__.py", line 364, in __init__

    self._handle = _dlopen(self._name, mode)

OSError: [WinError 193] %1 is not a valid Win32 application

Can someone help with this??

hkpeprah, 2020-05-26:
> Can someone help with this??
@jjaredrides I would open a new issue for different/new questions.  That being said, the issue is likely that you're using a 32-bit Python, but 64-bit application.  I would ensure you're using the 64-bit Python.