square/pylink #12: segmentation fault when unloading library on linux

charlesnicholson, 2018-01-03:
We observe regular segmentation faults on linux when exiting python that seem to be related to Library.unload(). It's not fully deterministic, but it appears much more regularly when the pylink.Library() object is created on a non-main thread.

I'm not sure that there's a 100% safe way to unload the dll on linux. I think it might be reasonable to get rid of the del and unload methods entirely, and simply leave the dll resident. Is there a use case for unloading and reloading the Library object and underlying dll?

Repro case:
```
> uname -a
Linux nicholsonc.<redacted>.google.com 4.9.0-4-amd64 #1 SMP Debian 4.9.65-3 (2017-12-03) x86_64 GNU/Linux

> python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)                          
[GCC 6.3.0 20170118] on linux                                          
Type "help", "copyright", "credits" or "license" for more information. 
>>> import pylink                                                      
>>> l = pylink.Library('../external/vendor/JLink/linux/libjlinkarm.so')
>>> j = pylink.JLink(l)                                                
>>> del j                                                              
>>> l.unload()                                                         
Segmentation fault                                                     
```

charlesnicholson, 2018-01-03:
Also:

```
Python 3.5.3 (default, Jan 19 2017, 14:11:04)                            
[GCC 6.3.0 20170118] on linux                                            
Type "help", "copyright", "credits" or "license" for more information.   
>>> import pylink                                                        
>>> l = pylink.Library('../external/vendor/JLink/linux/libjlinkarm.so')  
>>> j = pylink.JLink(l)                                                  
>>> quit()                                                               
Segmentation fault                                                       
```

charlesnicholson, 2018-01-03:
Without even instantiating a pylink.JLink:

```
Python 3.5.3 (default, Jan 19 2017, 14:11:04)                           
[GCC 6.3.0 20170118] on linux                                           
Type "help", "copyright", "credits" or "license" for more information.  
>>> import pylink                                                       
>>> l = pylink.Library('../external/vendor/JLink/linux/libjlinkarm.so') 
>>> l.unload()                                                          
Segmentation fault                                                      
```

hkpeprah, 2018-01-05:
Probably related to this: https://github.com/square/pylink/blob/master/pylink/library.py#L390
We could just use the Mac behaviour if that fixes it.  I was never entirely sold on this being the right way, but that's the behaviour I found when looking to what  the correct way of unloading is.

charlesnicholson, 2018-01-07:
I'll do some experiments and open a PR this week sometime.

My current line of thinking is:

Explicitly calling dlclose causes an immediate segfault, so we can't do that.

If we just release the variable reference to the .so (e.g. `self._lib = None`), that doesn't immediately unload it either; Python chooses when to do a GC. There are then 2 subsequent scenarios: A user creates a new Library object /before/ Python unloads the .so, or a user creates a new Library object /after/ Python unloads the .so.

I think either case is fine, since dlopen uses reference counting. If the .so is still resident when the user creates a new Library, the existing .so's refcount gets incremented and no heavy lifting is involved. If the .so is not resident, then it needs to be reloaded, which is also in line with the user's expectations.

The only case that isn't serviced here is forcing the explicit unloading and reloading of the .so file, meaning, maybe you need to reload the .so and recreate your Library and JLink objects without Python terminating. Maybe there's a new version of the SEGGER libraries that you want to deploy or something?

I certainly don't have that use case, FWIW, but if one really needed to do something like that, they'd be better serviced by deploying the new libraries to different locations on disk and not expecting to replace the existing library files while hot.

So, I think my CL is going to delete lines 389-391. I don't know what `del self._lib` does, but I'll read, experiment, and possibly add it.

WDYT?

hkpeprah, 2018-01-08:
Sounds fine to me.  I've only really ever tested fully on Mac/Windows, so Linux edge cases haven't been accounted for.  I would just delete the whole `else` block from `386 - 393`, and make the `Darwin` case the `else` case instead.  `library.load()` actually does copy the lib file to get around an early bug that existed in the DLL up until recently where you couldn't access more than one J-Link at a time.

hkpeprah, 2018-01-16:
Should be closed by PR #16, right?

charlesnicholson, 2018-01-16:
yup, thanks :)

closed 2018-01-16 by charlesnicholson