square/pylink #22: SWD/JTAG speed

mikolajrode, 2018-03-12:
- Maximum JTAG speed (MAX_JTAG_SPEED) is defined and this definition is used for any used interface (JTAG or SWD) instead of having it separated or having one universal interface speed definition.
- Maximum interface speed is actually dependent on JLink itself as in [https://www.segger.com/products/debug-probes/j-link/models/model-overview/](url). Currently 50 MHz is the fastest available.

hkpeprah, 2018-03-12:
Is the ask here to only check the speed against MIN/MAX when the interface is JTAG?

mikolajrode, 2018-03-12:
I'd refactor MAX_JTAG_SPEED into MAX_INTERFACE_SPEED (as it actually refers to SWD or JTAG) and push it to 50000 (50 MHz).
Second option, making less sense is to define separately MAX_JTAG_SPEED and MAX_SWD_SPEED and check them according to selected TIF (interface). However those max speeds would still be equal to 50 MHz.