square/pylink #29: flash_file() from GUI - WindowsError: exception: access violation writing 0x000000CE00000000

zamora18, 2018-04-19:
Hello, 

I have created a GUI (using PyQt4) that implements pylink flash_file() capability. It works great when developing in my IDE (PyCharm), but after bundling the GUI in an executable (using pyinstaller) I get this error (with varying locations 0x00____ ):

`WindowsError: exception: access violation writing 0x000000CE00000000`

This is the full error output:
```
Traceback (most recent call last):
  File "main_gui.py", line 87, in run
  File "main_gui.py", line 57, in _flash_binary
  File "build\bdist.win-amd64\egg\pylink\jlink.py", line 172, in wrapper
  File "build\bdist.win-amd64\egg\pylink\jlink.py", line 2021, in flash_file
WindowsError: exception: access violation writing 0x000000CE00000000
```
Also, it seems that the Segger J-flash/J-Link software is executed because this window pops up, but is stuck in this state:
[image]

Is there a way to show more detail on what is happening on the pylink side? The message does not seem very informative to me. Or could you inform me on what is happening and if there is a solution?

Thank you!

hkpeprah, 2018-04-19:
`WindowsError` generally means some windows issue with the system call.  I'd check that the DLL you're using corresponds to the version of the Python interpreter that you're running in.  After that, enabling logging is the best way to get more verbose output:
```
import logging
logging.basicConfig(level=logging.DEBUG)  # For maximum logging
```

sstallion, 2018-04-19:
Very cool! Would you be interested in contributing this as an example
application?

zamora18, 2018-04-19:
Just so you are aware, I edited my original post.
@hkpeprah Since the program performs the rest of the pylink functionality just fine (`pylink.JLink()`, `jlink.open()`, etc) I wouldn't think that the problem is with the DLL? Or am I wrong in thinking that?
I did not find any significant differences in the logging messages between the two, but maybe you see something I do not? I will include both below:
[gui_output.txt](https://github.com/square/pylink/files/1929652/gui_output.txt)
[pycharm_output.txt](https://github.com/square/pylink/files/1929653/pycharm_output.txt)
Are you familiar with [pyinstaller](https://www.pyinstaller.org/)? I am not very familiar, but in my research so far I haven't found anything that has helped me.
Would it be beneficial to send my code? 

@sstallion The GUI is a larger project, and the pylink functionality is only a part of that large project. I'm not sure it is something that would be very beneficial to contribute... However, I have created a [Very simple gui](https://user-images.githubusercontent.com/15371219/39016902-0424e18e-43df-11e8-967e-6e04c3f7ca0f.PNG) (in trying to isolate/debug this problem) that I would be more than happy to contribute (though it isn't anything special).

hkpeprah, 2018-04-19:
Huh, interesting.  So `pycharm` works, but the GUI fails?  Seems like it's failing on the line to download the file.  I've used `pyinstaller` before, and I know it creates some weird issues with some things.  I would attempt to try `flash()` instead of `flash_file()` (you'll have to read the file into a binary array), and it may help.

zamora18, 2018-04-19:
Correct, `pycharm` works but the GUI executable fails.
I tried `flash()` but unfortunately I am still having the same error.

sstallion, 2018-04-19:
Interesting. Is it possible that a different Python runtime is being used in the installer version?

zamora18, 2018-04-19:
It's possible, but I'm not sure that is what's happening. There is some [documentation](https://pythonhosted.org/PyInstaller/runtime-information.html) on run-time information for pyinstaller but I did not see anything relevant that page. I'm doing some reading and I'll look into that more.

zamora18, 2018-08-08:
I have attempted multiple solutions and still have the same problem. I created an issue in the pyinstaller repository, but they say [the problem is not on their end](https://github.com/pyinstaller/pyinstaller/issues/3576). 

After having done some research I believe the issue lies in how the binary is downloaded. Without understanding much of how it is implemented, it believe it currently requires accessing memory on the computer, which is denied because of DEP ([Data Execution Prevention](https://support.microsoft.com/en-us/help/875352/a-detailed-description-of-the-data-execution-prevention-dep-feature-in)) protection. Would there potentially be another way to download the file? Regardless of it being because of DEP, I do believe it is a problem with downloading (see log files below).

There was a suggested "fix" on the pyinstaller issue thread to run this command 
```python
os.system("start cmd /c JLinkSTM32.exe -SetDeviceFamily 2")
``` 
before the program crashes, but it has not worked for me. 

I will include the logs comparing my successful `flash_file()` with the failed attempts:

- [pycharm_output_blink_fast.txt](https://github.com/square/pylink/files/2271031/pycharm_output_blink_fast.txt)
    - success
- [pyinstaller_output_catch_exception_blink_fast.txt](https://github.com/square/pylink/files/2271033/pyinstaller_output_catch_exception_blink_fast.txt)
    - fail
- [pyinstaller_output_blink_fast.txt](https://github.com/square/pylink/files/2271034/pyinstaller_output_blink_fast.txt)
    - fail

I really appreciate any help you can offer.

zamora18, 2018-08-08:
I am not sure if this will fix _this_ issue (#29), but I will test the fix for #10

zamora18, 2018-08-08:
@hkpeprah That PR does fix my issue.

zamora18, 2018-08-10:
Update:
I re-attempted using both `flash_file()` and `flash()` methods.
- When using `flash_file()`, it crashes at the call:
    ```python
    bytes_flashed = self._dll.JLINK_DownloadFile(path.encode(), addr)
    ```
- When using `flash()`, it crashes at the call (within `flash_write()`):
    ```python
    bytes_flashed = self._dll.JLINKARM_EndDownload()
    ```
   - I found it interesting that it did not have any issues executing `self._dll.JLINKARM_BeginDownload(flags)` in `flash_write()`. 

Therefore I believe my issue relates to a problem with how the DLL downloads the binary files. 
Is there anything that can be done on the pylink end? Do you have any recommendations for me going forward?

zamora18, 2018-08-14:
After some final investigation I am closing this issue due to the problem most likely being due to DEP on Windows. 
My program executes successfully on Windows 7, but not any Windows 10 computers. I believe this is due to the changes in DEP for Windows since the release of Windows 7.

closed 2018-08-14 by zamora18