square/pylink #34: Cannot create JLink object on Windows

swohlerLL, 2018-05-30:
On Windows 10, using SEGGER Jlink tools v6.16d, using python 3.5.2 I cannot create a JLink object:

```
Python 3.5.2 |Anaconda 4.3.0 (64-bit)| (default, Jul  5 2016, 11:41:13) [MSC v.1900 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 5.1.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
More than one account associated with username

In [1]: import pylink

In [2]: jlink = pylink.JLink()
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-b073914a14e8> in <module>()
----> 1 jlink = pylink.JLink()

C:\Anaconda3\lib\site-packages\pylink\jlink.py in __init__(self, lib, log, detailed_log, error, warn, unsecure_hook)
    247
    248         if lib is None:
--> 249             lib = library.Library()
    250
    251         if lib.dll() is None:

C:\Anaconda3\lib\site-packages\pylink\library.py in __init__(self, dllpath)
    237             self.load(dllpath)
    238         else:
--> 239             self.load_default()
    240
    241     def __del__(self):

C:\Anaconda3\lib\site-packages\pylink\library.py in load_default(self)
    275
    276         if path is not None:
--> 277             return self.load(path)
    278
    279         return False

C:\Anaconda3\lib\site-packages\pylink\library.py in load(self, path)
    329
    330         self._temp = tf
--> 331         self._lib = ctypes.cdll.LoadLibrary(tf.name)
    332
    333         if self._windows:

C:\Anaconda3\lib\ctypes\__init__.py in LoadLibrary(self, name)
    423
    424     def LoadLibrary(self, name):
--> 425         return self._dlltype(name)
    426
    427 cdll = LibraryLoader(CDLL)

C:\Anaconda3\lib\ctypes\__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    345
    346         if handle is None:
--> 347             self._handle = _dlopen(self._name, mode)
    348         else:
    349             self._handle = handle

OSError: [WinError 193] %1 is not a valid Win32 application
```

I have copied the JLinkARM.dll and JLink_x64.dll files to the Windows/system32 directory and a user directory where the session is being run.

I have also attempted to pass in a path for the DLL:

```
In [3]: jlink = pylink.JLink('./JLink_x64.dll')
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-e018fd92e1f4> in <module>()
----> 1 jlink = pylink.JLink('./JLink_x64.dll')

C:\Anaconda3\lib\site-packages\pylink\jlink.py in __init__(self, lib, log, detailed_log, error, warn, unsecure_hook)
    249             lib = library.Library()
    250
--> 251         if lib.dll() is None:
    252             raise TypeError('Expected to be given a valid DLL.')
    253

AttributeError: 'str' object has no attribute 'dll'
```

And using ctypes to create a DLL object for the same:

```
In [4]: import ctypes

In [5]: jlink_dll = ctypes.WinDLL('./JLink_x64.dll')

In [6]: jlink = pylink.JLink(jlink_dll)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-067fffa0f666> in <module>()
----> 1 jlink = pylink.JLink(jlink_dll)

C:\Anaconda3\lib\site-packages\pylink\jlink.py in __init__(self, lib, log, detailed_log, error, warn, unsecure_hook)
    249             lib = library.Library()
    250
--> 251         if lib.dll() is None:
    252             raise TypeError('Expected to be given a valid DLL.')
    253

C:\Anaconda3\lib\ctypes\__init__.py in __getattr__(self, name)
    358         if name.startswith('__') and name.endswith('__'):
    359             raise AttributeError(name)
--> 360         func = self.__getitem__(name)
    361         setattr(self, name, func)
    362         return func

C:\Anaconda3\lib\ctypes\__init__.py in __getitem__(self, name_or_ordinal)
    363
    364     def __getitem__(self, name_or_ordinal):
--> 365         func = self._FuncPtr((name_or_ordinal, self))
    366         if not isinstance(name_or_ordinal, int):
    367             func.__name__ = name_or_ordinal

AttributeError: function 'dll' not found
```

I assume there is some issue with the library related to 32/64-bit support, or perhaps it is not supported in python 3.5.

swohlerLL, 2018-05-30:
I was able to instantiate an object by creating a library object from the DLL:

```
In [7]: lib = pylink.library.Library('./JLink_x64.dll')

In [8]: lib
Out[8]: <pylink.library.Library at 0x1bdd06e6588>

In [9]: jlink = pylink.JLink(lib)

In [10]: jlink
Out[10]: <pylink.jlink.JLink at 0x1bdd06ebda0>

In [11]: jlink.version
Out[11]: '6.16d'
```

hkpeprah, 2018-05-30:
This should be better documented, I apologize.  This is the open 64-bit Windows issue.  I will look again at the open PR for this, and try to address the issues I initially had.

swohlerLL, 2018-05-30:
No worries, I see the PR now. Wasn't trying to be overly critical. Thanks for the prompt response!

closed 2018-05-30 by swohlerLL