square/pylink #35: GPIO Control

henryay1, 2018-05-31:
Hello,

I'm trying to control the GPIOs on the JLink but noticed that JLINK_EMU_GPIO_SetState has been obsoleted in the SDK which makes jlink.gpio_set()'s implementation obsolete as well. As a workaround, I'm trying to use the JLINK_IFUNC_PIN_OVERRIDE function as mentioned in the SDK user guide. They provide an example in the guide and I tried implementing the example in Python using ctypes, but I cannot figure out how to successfully implement this in Python. I do not see any code in Pylink that uses these JLINK_IFUNC functions, so, I'm a little stuck. Can someone suggest a way to implement this or point me in the right direction?

Thanks,
Henry

CC: @charlesnicholson

charlesnicholson, 2018-06-04:
@hkpeprah @sstallion any insights here? Henry's working with us; if you could give us any pointers we'd appreciate it!

hkpeprah, 2018-06-04:
Sorry, meant to reply to this over the weekend.  Unfortunately, I've never used that API.  I'll take a look later to see.  What issue are you running into?

henryay1, 2018-06-05:
I'm a newbie to working with ctypes for more advanced uses in Python, so, I'm just having trouble figuring out the best way of translating the example given in the C example in the user guide to Python. There aren't too many examples on the web on casting a function prototype to a pointer (I think that's what the example is). I tried a few methods of implementing it, but I keep getting segfaults or implementation errors.

hkpeprah, 2018-06-05:
There's an example of binding a callback here: https://github.com/square/pylink/blob/master/pylink/jlink.py#L415

Note that we have to create a `ctype` function to indicate what the function is: `LOG_PROTOTYPE = ctypes.CFUNCTYPE(None, ctypes.c_char_p)`; this basically specifies the return type, and any number of arguments that the function takes.  In this instance, callback will be called with a string (string pointer), and does not return anything (is `void`).

More information about `CFUNCTYPE`s can be found here: https://docs.python.org/2/library/ctypes.html#callback-functions

willyteske, 2023-02-14:
Hello,

I am interested in using the PIN Override functionality as well. I was able to use the `JLINK_FUNC_PIN_OVERRIDE` function that is exposed via `JLINK_GetpFunc(...)`.

See [jlink_ifunc_pin_override.py.txt](https://github.com/square/pylink/files/10732614/jlink_ifunc_pin_override.py.txt)

Shell I create a pull request based on this POC?

hkpeprah, 2023-02-14:
Feel free to create a PR; submissions are always welcome.