square/pylink #36: add support for remote connections in tunneling mode

Sauci, 2018-07-04:
SEGGER provides the ability to [connect to a device through a free tunneling server](https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/), which would be a pretty ~~cool~~ useful feature.
I would suggest to update the ```JLink.open``` method to provide the possibility to call it like this: 
```python
JLink.open(self, ip_addr='tunnel', serial_no=123456789)
```
this modification would keep the backward compatibility with older versions of this package.

hkpeprah, 2018-07-05:
This is currently not work?  `open()` does take a `serial_no` and `ip_addr` parameter.

Sauci, 2018-07-05:
no, because to use this feature, the call to dll function ```JLINKARM_SelectIP``` should be
```python
JLINKARM_SelectIP('tunnel' + str(serial_no), 19020)
```

as the current API is
```python
if ip_addr is not None:
    addr, port = ip_addr.split(':')
    if serial_no is None:
        result = self._dll.JLINKARM_SelectIP(addr.encode(), int(port))
        if result == 1:
            raise errors.JLinkException('Could not connect to emulator at %s.' % ip_addr)
        else:
            # Note: No return code when selecting IP by serial number.
            self._dll.JLINKARM_EMU_SelectIPBySN(int(serial_no))
```
executing ```JLink.open(self, ip_addr='tunnel', serial_no=123456789)``` will crash in the ```split(':')``` and ```JLink.open(self, ip_addr='tunnel:19020', serial_no=123456789)``` will call ```JLINKARM_EMU_SelectIPBySN``` instead of ```JLINKARM_SelectIP```.

hkpeprah, 2018-07-05:
Ah, okay.  This sounds like the `ip_addr` should be `tunnel:123456789:19020`, and the `split` is the issue.  Seems like what we should be doing is:
```
addr, port = ip_addr.rsplit(':', 1)
```
```
>>> addr, port = 'tunnel:123456789:19020'.rsplit(':', 1)
>>> print(addr)
tunnel:123456789
>>> print(port)
19020
```
I'm open to a pull-request, otherwise I can put one up later in the week.

Sauci, 2018-07-05:
yes, it would work. but don't you think it would be less confusing to keep providing the serial number  trough the ```serial_no``` keyword argument?

hkpeprah, 2018-07-05:
It would.  My hesitation is that we'd be treating `tunnel` as a reserved keyword, such that we'd have something like:
```
addr, port = ip_addr.split(':')
if addr == 'tunnel' and serial_no:
    addr += ':%s' % (serial_no)
    # Connect here...
elif serial_no is None:
    pass
else:
    pass
```

Sauci, 2018-07-05:
looks good! a last question, should we raise an ValueError if the ip address is ```tunnel:1234``` and the ```serial_no=None```? I guess it would make sense, as the serial number is a required argument in that case.

hkpeprah, 2018-07-20:
Should be available in the latest release.  Thanks for the patch!

closed 2018-09-04 by Sauci