square/pylink #38: Why does it appear that 'pylink.errors.JLinkException: Unsupported device was connected to.'

zengjiawei, 2018-09-03:
``
from pylink import jlink

dllpath = 'C:\SoftWare\Office SoftWare\keil5.14\SEGGER\JLink_V614b\JLinkARM.dll'

def start():
    jlk = jlink.JLink(lib=jlink.library.Library(dllpath=dllpath))
    jlk.open(serial_no=123456)
    print(jlk.product_name)
    jlk.oem
    print(jlk.opened())
    print(jlk.connected())
    print(jlk.target_connected())
    print(jlk.update_firmware())

    jlk.connect(chip_name='chip')

    print(jlk.core_id())

if __name__ == '__main__':
    start()
``

zengjiawei, 2018-09-04:
I've solved it, Thanks

closed 2018-09-04 by zengjiawei

Stimaleger, 2018-11-16:
I get the same error, can you tell me how you solve it plz ?

zengjiawei, 2018-11-16:
You need to specifies the path of the DLL

zeng_jiawei@163.com
 
From: Stimaleger
Date: 2018-11-16 20:45
To: square/pylink
CC: zengjiawei; State change
Subject: Re: [square/pylink] Why does it appear that 'pylink.errors.JLinkException: Unsupported device was connected to.' (#38)
I get the same error, can you tell me how you solve it plz ?

Stimaleger, 2018-11-16:
Thx for your fast answer ! 
When I call the open method I should pass the path of the jlink dll ?

Stimaleger, 2018-11-16:
Thx for the fast answer ?
The example you gave above is working ?

zengjiawei, 2018-11-16:
Yes, officials did not pointed out , but i found it will appear error  what you said. you can try latter.

zeng_jiawei@163.com
 
From: Stimaleger
Date: 2018-11-16 21:14
To: square/pylink
CC: zengjiawei; State change
Subject: Re: [square/pylink] Why does it appear that 'pylink.errors.JLinkException: Unsupported device was connected to.' (#38)
Thx for your fast answer !
When I call the open method I should pass the path of the jlink dll ?

Stimaleger, 2018-11-16:
Thx it works.