square/pylink #41: Disable popups

johncharlie, 2018-09-06:
When using an STM32 board as a JLink device, every day a popup appears so you cannot execute scripts in an automated manner, because you need to close the window. Is it possible to disable all the popups?

hkpeprah, 2018-09-06:
After you create the instance, you can can call `.disable_dialog_boxes()`.  This has the effect of disabling the boxes that pop-up for FW update, and switches it to batch mode so that you don't get prompted.
```
jlk = pylink.JLink()
jlk.disable_dialog_boxes()
<rest of stuff>
```

johncharlie, 2018-09-06:
Thanks, will test and update the issue if it works for all popups.

johncharlie, 2018-09-07:
It does not work with this popup:

[image]

hkpeprah, 2018-09-07:
Hm.  That seems to not be dismissible (only some are).  Given the box that says "Do not show this message again for today" and the terms of use, I don't think it's intended for batch mode.

sstallion, 2018-09-07:
> Yes, I will need to do that once every day, the only problem is that my
> code runs automated. Thanks.

It might be worth opening a support ticket with SEGGER to see if there is
another method you can use to silence those popups - that's awfully
frustrating.

johncharlie, 2018-09-07:
Will do. Thanks.

johncharlie, 2018-09-27:
After talking with SEGGER support, only way to suppress this popup is by using a JLink device and connecting it to the STLink board. The popup cannot be disabled since this support (converting STLink device as JLink device) is only for evaluation purposes. Thanks, I will close the issue.

closed 2018-09-27 by johncharlie

sstallion, 2018-09-27:
Thanks Juan, I'm sorry there isn't something more we can do.