square/pylink #45: Windows10 pylink.JLink() fail

zhaojuntao, 2018-12-07:
Windows10 pylink.JLink() fail:

```
Traceback (most recent call last):
  File "test.py", line 8, in <module>
    jlink = pylink.JLink()
  File "C:\Python37\lib\site-packages\pylink_square-0.1.2-py3.7.egg\pylink\jlink.py", line 249, in __init__
  File "C:\Python37\lib\site-packages\pylink_square-0.1.2-py3.7.egg\pylink\library.py", line 239, in __init__
  File "C:\Python37\lib\site-packages\pylink_square-0.1.2-py3.7.egg\pylink\library.py", line 277, in load_default
  File "C:\Python37\lib\site-packages\pylink_square-0.1.2-py3.7.egg\pylink\library.py", line 331, in load
  File "C:\Python37\lib\ctypes\__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python37\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 不是有效的 Win32 应用程序。
```

MJBina, 2018-12-18:
I am having the same issue. 
It appears to be an issue with 64-bit Python and 32-bit PyLink.  Correct?

I am using SEGGER 6.40.  I noticed that there is a 64-bit JLink .DLL `JLink_x64.dll`.  Could this help?
I also noticed that SEGGER has a Beta version (v6.41a) of 64-Bit JLink (link below)  Has anyone tried this?
[https://www.segger.com/downloads/jlink/#J-LinkSoftwareAndDocumentationPackBeta](url)
```
Traceback (most recent call last):
  File "C:\projects\piranha\assembly_tester\dev\python\test_pylink.py", line 6, in <module>
    jlink = pylink.JLink()
  File "C:\Program Files\Python37\lib\site-packages\pylink\jlink.py", line 249, in __init__
    lib = library.Library()
  File "C:\Program Files\Python37\lib\site-packages\pylink\library.py", line 239, in __init__
    self.load_default()
  File "C:\Program Files\Python37\lib\site-packages\pylink\library.py", line 277, in load_default
    return self.load(path)
  File "C:\Program Files\Python37\lib\site-packages\pylink\library.py", line 331, in load
    self._lib = ctypes.cdll.LoadLibrary(tf.name)
  File "C:\Program Files\Python37\lib\ctypes\__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "C:\Program Files\Python37\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
```

My Environment - has problem

```
Microsoft Windows [Version 10.0.14393]
(c) 2016 Microsoft Corporation. All rights reserved.

C:\>python
Python 3.7.1 (v3.7.1:260ec2c36a, Oct 20 2018, 14:57:15) [MSC v.1915 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

hkpeprah, 2018-12-19:
This is a known bug.  I just merged in a PR #25, which should remedy this issue for you if you upgrade to `0.1.3` of the package.

zhaojuntao, 2018-12-20:
@hkpeprah Thanks

ReddyKumarUppu, 2020-12-29:
I am not able to connect to ATSAMV70Q20 device.