square/pylink #46: provide an enum for registers

Sauci, 2018-12-11:
Hi,

I would like to know if it would be possible to provide an enumeration to clarify the calls to ```register_read(self, register_index)```. I guess that calling this function with ```register_index = 0``` the value of register **r0** will be returned, but I'm not sure what index should be used for **CPS** register, more generally what index should be used for banked registers. I guess that an enum like shown bellow might help the users:

```
REG_R0 = 0
...
REG_R7 = 7
REG_R8_USR = ?
REG_R8_FIQ = ?
REG_R9_USR = ?
REG_R9_FIQ = ?
```

I know it is possible to read the banked register **r8_fiq** with GDB, even if the controller is not in **FIQ** mode, but I don't know if it would be possible using the DLL. 

[here](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0460d/ch03s06s01.html)'s a link showing different registers available for a **cortex-r5** controller (only as example).

Thanks for your feedback

hkpeprah, 2018-12-11:
Sounds like a good idea / enhancement.  Something like a registers `enum` with
```
class Registers(enum):
    ARM_CORTEX_R5_R0 = 0
    <...>
    ARM_CORTEX_R5_CPS
```

sstallion, 2018-12-11:
Does the JLink API provide anything similar?

hkpeprah, 2018-12-11:
There's the `register_name()` method which takes an index and returns the name of the register at that index.  There's also `register_list()` which will return a list of the register numbers.  Together you could probably do something like:

```
register_map = {}
for reg_index in jlink.register_list():
    register_map[jlink.register_name(reg_index)] = reg_index
print(jlink.register_read(register_map['<REGISTER_NAME>']))
```

Sauci, 2018-12-11:
Oh sorry I didn’t see those two functions...
I guess my proposal doesn’t make sense anymore, as the registers are known at runtime. I was thinking that it would be great for auto completion, but it doesn’t seems to be possible.
Should I close the issue?

Sorry again, RTFD the next time

hkpeprah, 2019-09-09:
Ah, super sorry about getting back to you late on this.  I think probably using the `list` is the best bet.  It might be worth modifying the register commands to support both named and indexed registers if that's something you want to take a look at.

Sauci, 2019-09-11:
No problem. Ok, I will add this feature. What do you think about this proposition:
- modifying ```register_read(self, register_index)``` signature to 
  ```register_read(self, register_index=None, register_name=None) ```
  This change would be backward compatible.

- modifying ```register_write(self, reg_index, value)``` signature to
  ```register_write(self, value, reg_index=None, register_name=None)```
  This change wouldn't be backward compatible.

This has the downside to break backward compatibility with current/previous versions of the module, but it keeps a clear interface. Otherwise, if you prefer to keep backward compatibility, I could add functions like ```read_register_by_name``` and ```write_register_by_name```.

hkpeprah, 2019-09-23:
Release with patch should be available in v0.3.0: https://github.com/square/pylink/releases/tag/v0.3.0

Sauci, 2019-09-24:
Great, thanks! I guess this issue can be closed?

hkpeprah, 2019-09-24:
Sounds good.

closed 2019-09-24 by hkpeprah