square/pylink #47: Calling pylink command line causes confusing error

samvasko, 2018-12-13:
Looks like the tool assumes that the argument is always provided. Its bit difficult to track down why the error is shown.

```
$ pylink
Traceback (most recent call last):
  File "/Users/sam/.local/share/virtualenvs/pylink-d8aVmqP2/bin/pylink", line 11, in <module>
    load_entry_point('pylink-square', 'console_scripts', 'pylink')()
  File "/Users/sam/Code/entia/aptus/pylink/pylink/__main__.py", line 598, in main
    args.command(args)
AttributeError: 'Namespace' object has no attribute 'command'
```

hkpeprah, 2018-12-13:
Hm, I get a different message when I do this:
```
(riker) [ford@sqr12850] /Users/ford/dev/square/riker $ pylink
usage: pylink [-h] [--version] [-v]
              {emulator,info,firmware,flash,unlock,erase,license} ...
pylink: error: too few arguments
```
Is this on the latest version?

kylemanna, 2021-09-29:
This seems to have popped up again on the latest `0.11.1` release on Arch Linux running Python 3.9.7.  Fails in an AUR package build chroot, fails on my host system, and fails in a `venv` with `requirements.txt` installed.

```
~/t/r/pylink ❯❯❯ python3 -m venv venv
~/t/r/pylink ❯❯❯ source venv/bin/activate
(venv) ~/t/r/pylink ❯❯❯ pip install -r requirements.txt
Collecting behave==1.2.5
  Downloading behave-1.2.5-py2.py3-none-any.whl (109 kB)
     |████████████████████████████████| 109 kB 8.6 MB/s 
Collecting coverage==4.4.1
  Downloading coverage-4.4.1.tar.gz (369 kB)
     |████████████████████████████████| 369 kB 59.6 MB/s 
Collecting future
  Downloading future-0.18.2.tar.gz (829 kB)
     |████████████████████████████████| 829 kB 135.9 MB/s 
Collecting psutil>=5.2.2
  Downloading psutil-5.8.0-cp39-cp39-manylinux2010_x86_64.whl (293 kB)
     |████████████████████████████████| 293 kB 16.5 MB/s 
Collecting pycodestyle>=2.3.1
  Downloading pycodestyle-2.7.0-py2.py3-none-any.whl (41 kB)
     |████████████████████████████████| 41 kB 951 kB/s 
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting sphinx==1.4.8
  Downloading Sphinx-1.4.8-py2.py3-none-any.whl (1.6 MB)
     |████████████████████████████████| 1.6 MB 100.9 MB/s 
Collecting sphinx-argparse==0.1.15
  Downloading sphinx-argparse-0.1.15.tar.gz (6.0 kB)
Collecting sphinx_rtd_theme==0.2.4
  Downloading sphinx_rtd_theme-0.2.4-py2.py3-none-any.whl (1.4 MB)
     |████████████████████████████████| 1.4 MB 62.1 MB/s 
Collecting sphinxcontrib-napoleon==0.5.3
  Downloading sphinxcontrib_napoleon-0.5.3-py2.py3-none-any.whl (24 kB)
Collecting wheel
  Downloading wheel-0.37.0-py2.py3-none-any.whl (35 kB)
Collecting parse>=1.6.3
  Downloading parse-1.19.0.tar.gz (30 kB)
Collecting parse-type>=0.3.4
  Downloading parse_type-0.5.2-py2.py3-none-any.whl (32 kB)
Collecting docutils>=0.11
  Downloading docutils-0.17.1-py2.py3-none-any.whl (575 kB)
     |████████████████████████████████| 575 kB 41.4 MB/s 
Collecting snowballstemmer>=1.1
  Downloading snowballstemmer-2.1.0-py2.py3-none-any.whl (93 kB)
     |████████████████████████████████| 93 kB 2.4 MB/s 
Collecting Jinja2>=2.3
  Downloading Jinja2-3.0.1-py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 98.1 MB/s 
Collecting Pygments>=2.0
  Downloading Pygments-2.10.0-py3-none-any.whl (1.0 MB)
     |████████████████████████████████| 1.0 MB 128.2 MB/s 
Collecting babel!=2.0,>=1.3
  Downloading Babel-2.9.1-py2.py3-none-any.whl (8.8 MB)
     |████████████████████████████████| 8.8 MB 93.4 MB/s 
Collecting imagesize
  Downloading imagesize-1.2.0-py2.py3-none-any.whl (4.8 kB)
Collecting alabaster<0.8,>=0.7
  Downloading alabaster-0.7.12-py2.py3-none-any.whl (14 kB)
Collecting pockets>=0.3
  Downloading pockets-0.9.1-py2.py3-none-any.whl (26 kB)
Collecting pytz>=2015.7
  Downloading pytz-2021.1-py2.py3-none-any.whl (510 kB)
     |████████████████████████████████| 510 kB 123.8 MB/s 
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.0.1-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (30 kB)
Using legacy 'setup.py install' for coverage, since package 'wheel' is not installed.
Using legacy 'setup.py install' for sphinx-argparse, since package 'wheel' is not installed.
Using legacy 'setup.py install' for future, since package 'wheel' is not installed.
Using legacy 'setup.py install' for parse, since package 'wheel' is not installed.
Installing collected packages: pytz, MarkupSafe, snowballstemmer, six, Pygments, parse, Jinja2, imagesize, docutils, babel, alabaster, sphinx, pockets, parse-type, wheel, sphinxcontrib-napoleon, sphinx-rtd-theme, sphinx-argparse, pycodestyle, psutil, future, coverage, behave
    Running setup.py install for parse ... done
    Running setup.py install for sphinx-argparse ... done
    Running setup.py install for future ... done
    Running setup.py install for coverage ... done
Successfully installed Jinja2-3.0.1 MarkupSafe-2.0.1 Pygments-2.10.0 alabaster-0.7.12 babel-2.9.1 behave-1.2.5 coverage-4.4.1 docutils-0.17.1 future-0.18.2 imagesize-1.2.0 parse-1.19.0 parse-type-0.5.2 pockets-0.9.1 psutil-5.8.0 pycodestyle-2.7.0 pytz-2021.1 six-1.16.0 snowballstemmer-2.1.0 sphinx-1.4.8 sphinx-argparse-0.1.15 sphinx-rtd-theme-0.2.4 sphinxcontrib-napoleon-0.5.3 wheel-0.37.0
WARNING: You are using pip version 21.2.3; however, version 21.2.4 is available.
You should consider upgrading via the '/home/nitro/tmp/repos/pylink/venv/bin/python3 -m pip install --upgrade pip' command.
(venv) ~/t/r/pylink ❯❯❯ python3 -m pylink
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/nitro/tmp/repos/pylink/pylink/__main__.py", line 608, in <module>
    exit(main())
  File "/home/nitro/tmp/repos/pylink/pylink/__main__.py", line 599, in main
    args.command(args)
AttributeError: 'Namespace' object has no attribute 'command'
(venv) ~/t/r/pylink ❯❯❯ python3 --version
Python 3.9.7
(venv) ~/t/r/pylink ❯❯❯ python --version
Python 3.9.7
```

Haven't had a chance to dig in to it yet, but seems to be an issue with Python 2/3 argparse as Python 2.7.18 works fine:
```
(venv) ~/t/r/pylink ❯❯❯ python2 --version
Python 2.7.18
(venv) ~/t/r/pylink ❯❯❯ python2 -m pylink --version
pylink 0.11.1
```

closed 2021-10-05 by hkpeprah