square/pylink #48: how to get RTT buffer size?

tianxiaoMCU, 2019-01-24:
The J-Link RTT Viewer can get channel description include buffer size,like below
[image]
Is there a API to do this? I try rtt_control,but failed.

hkpeprah, 2019-01-24:
Doesn't seem like we expose this yet.  `rtt_control` is the method to use, I believe.

sstallion, 2019-01-24:
It might be worth scoping this out and creating a ticket.

tarakanov, 2019-03-04:
+1 for this feature. I use RTT for debugging and cannot send strings more than 14 characters long (for 16 bytes down buffer)

closed 2020-03-19 by hkpeprah

hkpeprah, 2020-03-19:
Should be available now in [v0.6.0](https://github.com/square/pylink/releases/tag/0.6.0).

PR #52 (2020-03-19): [Feature] RTT Control Buffer and Status
## Feature: RTT Control Buffer and Status
### Description
This PR fleshes out the RTT support to resolve #48 and #51.  Specifically, this PR adds the ability to pass an address for the RTT control block when starting RTT, get the current RTT status (including number of up and down buffers), and get the descriptor for an RTT buffer (which returns information such as the name of the buffer, and the size of the buffer).

### Testing
A new firmware image has been added that builds in the RTT vendor code and implements a simple echo program.  The firmware rea [... cut]
commit d2e761da (2020-03-19): [Feature] RTT Control Buffer and Status (#52)

* Feature: RTT Control for Buffer Description and Status Information

Adds support via `rtt_get_status()` and `rtt_get_buf_descriptor()` to enable
querying for RTT status and buffer information respectively.  Both features
contribute to and resolve #48.

Adds an optional `block_address` parameter to `rtt_start()` to enable
configuring the address for the RTT control block to support devices where the
control block cannot be found automatically.  This resolves #48.

* Support: Functional test for RTT Support

Adds a functional test for RTT support. [... cut]