square/pylink #49: target_connected() return error status

tianxiaoMCU, 2019-01-28:
1. At the beginning，target_connected() return True；
2. then disconnect all cables between jlink and target（connection between PC and JLink still exist）；
3. target_connected() **still return True**；

closed 2019-01-28 by tianxiaoMCU

hkpeprah, 2019-01-28:
I think we'd have to try a debugging command of some sort (that would always succeed if the target is connected), and check that it fails in order to determine if the cable was un-plugged, as if the method returns true, it believes the target is still connected.

JeremiahGillis, 2021-06-22:
I was surprised when `target_connected()` always returned `True` upon a J-Link cable disconnect or when the device was unplugged. I wonder how Segger intended this to work...

What I did, which only works with self powered device (not powered via J-Link), was to monitor `VTarget` in `hardware_status`. I have a worker thread watch this and trigger a target disconnect when the voltage is < 0.25 V. I haven't had any issues with this method.