square/pylink #54: num_supported_devices in connect() adds test time

wallacbe, 2019-05-03:
num_supported_devices() call inconnect adds > 100 ms test time each call.  If we are not using lock/unlock can we add flag argument to skip?

https://github.com/square/pylink/blob/master/pylink/jlink.py#L933

hkpeprah, 2019-05-06:
I would rather not add a flag for this, but I think making `_device` be a property on doing it when needed makes sense, and would produce the same result.  E.g.

```
def connect(...):
    <snipped>
    self._chip_name = chip_name

def disconnect(...):
    self._device = None

@property
def device(self):
    if self._device:
        return self._device
    <do lookup>
    return instance
```

If you want to go ahead and submit a patch, I'd be more than happy to merge it.

hkpeprah, 2024-07-20:
This has been resolved with #74 .

closed 2024-07-20 by hkpeprah