square/pylink #58: add support Segger tunnel server

Sauci, 2019-09-06:
I'm not sure if it would be possible, but it would be great to be able to open a device over the Segger's remote server using the **name**/**serial number** and optional **password** parameters.

To launch the server, provide either the serial number or the name, and an optional password:
- ```./JLinkRemoteServerCLExe -UseTunnel -TunnelBySN <serial_number> -TunnelPW <password>```
- ```./JLinkRemoteServerCLExe -UseTunnel -TunnelByName <name> -TunnelPW <password>```

Once the server is running, it is possible to connect to it using **JLinkExe** application:
- ```JLinkExe ip tunnel:<serial_number>:<password>```
- ```JLinkExe ip tunnel:<tunnel_name>:<password>```

Does anyone knows if this feature is available in the latest version of the DLL? My first guess would be to call the DLL function ```JLINKARM_SelectIP``` with ```"tunnel:<name>:<password>"```, but I don't know how to handle the second **port** parameter.

hkpeprah, 2019-09-09:
Someone added [`open_tunnel()`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.open_tunnel), which takes a serial number and port; I suppose you could pass in the tunnel name in the place of the serial number, and password in place of the port, and it might work?

Sauci, 2019-09-11:
Yes, I've added this feature (issue #36).
When we call ```open_tunnel()```, the function internally calls ```open()``` and ```open()``` calls the JLink API ```JLINKARM_SelectIP(address=str, port=int)```. 
Internally, both ```address``` and ```port``` parameters are extracted from the ```ip_addr``` parameter of the ```open``` function using the ```rsplit``` function (see [file changes](https://github.com/square/pylink/pull/37/files) of issue #36).
However, if the ```open()``` function if called with ```ip_addr=tunnel:<serial_number>:<password>```, there is no way to retrieve the integer value of ```port```, which is required in the call of ```JLINKARM_SelectIP``` API.

hkpeprah, 2024-07-20:
Should be resolved by #36 .

closed 2024-07-20 by hkpeprah