square/pylink #66: how to download the elf file

DerZc, 2019-12-20:
very sorry for interrupting you.
can you tell me how to download the elf file?
i can't use the flash() or flash_file() to download the elf file.

DerZc, 2019-12-22:
I convert the elf file to bin file and download it to the board.
But I run the trace example, and set the trace_address as the entry of main, set the breakpoint_address as a point in the main function. But I don't get any instruction. I don't get any error. How can I get the instruction in the trace.

DerZc, 2019-12-22:
And I meet the question that "No handlers could be found for logger "pylink.jlink"", how can I do?
Thank you very much.

hkpeprah, 2019-12-23:
Sorry for getting back to you so late.  You need to great a bin (or hex) file from the ELF file to flash to your device.  The no handlers message that pops up means that the library attempted to log a message (could be informational, error, etc.), but there was no handler to do so.  To enable logging, you should do something like:
```
import logging

logging.basicConfig(level=logging.INFO)
```
or `logging.DEBUG`, `logging.ERROR`, etc. depending on what logging level you want.

DerZc, 2019-12-23:
I get it. Thank you for your help.
I  run the trace example but there is no instructions in trace, how can I do?

hkpeprah, 2019-12-23:
The example should work provided the addresses are valid (come from the disassembly of the binary file).  That would be my suggestion, otherwise I'm unsure of what the issue would be.  I'd ensure that the device you're using supports instruction tracing as well (the info command from the CLI will tell you that: `pylink info -p`).

DerZc, 2019-12-23:
Thank you for your help. May be the address I set was wrong, I reset the address and it run success.
Now I can get the trace when it hit the breakpoint, but how can I get the instructions when it is running.

DerZc, 2019-12-23:
I'm so sorry that I have so much question.
I run the Strace example to get the trace, but the example do not return all the address in the trace.
the trace I want is
` OS_TASK_CREATE(&TCBHP, "HP Task", 100, HPTask, StackHP);
     0800 01B8   MOVS         R3, #2
     0800 01BA   STR          R3, [SP, #8]
     0800 01BC   MOV.W        R3, #0x200
     0800 01C0   STR          R3, [SP, #4]
     0800 01C2   LDR          R3, =StackHP               ; [0x08000240] =0x20000848
     0800 01C4   STR          R3, [SP]
     0800 01C6   LDR          R3, =0x08000269            ; [0x08000244]
     0800 01C8   MOVS         R2, #0x64
     0800 01CA   LDR          R1, =0x08004D90            ; [0x08000248]
     0800 01CC   LDR          R0, =TCBHP                 ; [0x0800024C] =0x20000EF8
     0800 01CE   BL           OS_TASK_Create_DP          ; 0x080029CC`
and the STrace example is:
`import pylink

import sys
import time

import logging

def strace(device, trace_address, breakpoint_address):
    """Implements simple trace using the STrace API.
    Args:
      device (str): the device to connect to
      trace_address (int): address to begin tracing from
      breakpoint_address (int): address to breakpoint at
    Returns:
      ``None``
    """
    logging.basicConfig(level=logging.INFO)
    jlink = pylink.JLink()
    jlink.open('932000223')

    # Do the initial connection sequence.
    jlink.power_on()
    jlink.set_tif(pylink.JLinkInterfaces.SWD)
    jlink.connect(device)
    jlink.flash_file('./Start.bin', 0x8000000)
    jlink.reset()

    # Clear any breakpoints that may exist as of now.
    jlink.breakpoint_clear_all()

    # Start the simple trace.
    op = pylink.JLinkStraceOperation.TRACE_START
    jlink.strace_clear_all()
    jlink.strace_start()

    # Set the breakpoint and trace events, then restart the CPU so that it
    # will execute.
    bphandle = jlink.breakpoint_set(breakpoint_address, thumb=True)
    trhandle = jlink.strace_code_fetch_event(op, address=trace_address)
    jlink.restart()
    time.sleep(1)
    # jlink.halt()
    f = open("trace.txt", "w") 

    # Run until the CPU halts due to the breakpoint being hit.
    while True:
        if jlink.halted():
            break
        #print('%#x'%jlink.register_read(15), file = f)
        # print >> f, '%#x'%jlink.register_read(15)
        # jlink.step()
        # if jlink.register_read(15) == 0x080001f2:
        #     break;
        
    
    # Print out all instructions that were captured by the trace.
    while True:
        instructions = jlink.strace_read(0x10000)
        if len(instructions) == 0:
            break
        instruction = instructions[0]
        # print(jlink.disassemble_instruction(instruction))
        for i in instructions:
            print('%#x'%i)
        # print >> f, jlink.disassemble_instruction(instruction)

    f.close()
    jlink.power_off()
    jlink.close()

if __name__ == '__main__':
    exit(strace('STM32F407VE', 0x080001b8, 0x080001ce))`
the result I got is:
`INFO:pylink.jlink:Device "STM32F407VE" selected.
INFO:pylink.jlink:Found SW-DP with ID 0x2BA01477
INFO:pylink.jlink:Found SW-DP with ID 0x2BA01477
INFO:pylink.jlink:Scanning AP map to find all available APs
INFO:pylink.jlink:AP[1]: Stopped AP scan as end of AP map has been reached
INFO:pylink.jlink:AP[0]: AHB-AP (IDR: 0x24770011)
INFO:pylink.jlink:Iterating through AP map to find AHB-AP to use
INFO:pylink.jlink:AP[0]: Core found
INFO:pylink.jlink:AP[0]: AHB-AP ROM base: 0xE00FF000
INFO:pylink.jlink:CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
INFO:pylink.jlink:Found Cortex-M4 r0p1, Little endian.
INFO:pylink.jlink:FPUnit: 6 code (BP) slots and 2 literal slots
INFO:pylink.jlink:CoreSight components:
INFO:pylink.jlink:ROMTbl[0] @ E00FF000
INFO:pylink.jlink:ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
INFO:pylink.jlink:ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
INFO:pylink.jlink:ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
INFO:pylink.jlink:ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
INFO:pylink.jlink:ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
INFO:pylink.jlink:ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
INFO:pylink.jlink:Found SW-DP with ID 0x2BA01477
INFO:pylink.jlink:Found SW-DP with ID 0x2BA01477
INFO:pylink.jlink:AP map detection skipped. Manually configured AP map found.
INFO:pylink.jlink:AP[0]: AHB-AP (IDR: Not set)
INFO:pylink.jlink:AP[0]: Core found
INFO:pylink.jlink:AP[0]: AHB-AP ROM base: 0xE00FF000
INFO:pylink.jlink:CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
INFO:pylink.jlink:Found Cortex-M4 r0p1, Little endian.
INFO:pylink.jlink:FPUnit: 6 code (BP) slots and 2 literal slots
INFO:pylink.jlink:CoreSight components:
INFO:pylink.jlink:ROMTbl[0] @ E00FF000
INFO:pylink.jlink:ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
INFO:pylink.jlink:ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
INFO:pylink.jlink:ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
INFO:pylink.jlink:ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
INFO:pylink.jlink:ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
INFO:pylink.jlink:ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
INFO:pylink.jlink:J-Link: Flash download: Bank 0 @ 0x08000000: Skipped. Contents already match
INFO:pylink.jlink:Reset: Halt core after reset via DEMCR.VC_CORERESET.
INFO:pylink.jlink:Reset: Reset device via AIRCR.SYSRESETREQ.
0x80001cc
0x80001ca
0x80001c8
0x80001c6
0x80001c4
0x80001c2
0x80001c0
0x80001bc
`
there are two address was missed:
`0x80001b8` and `0x80001ba`
I got other trace, sometime it is completed, but sometime some address was missed.
the project I run was come from the jlink website:
[J-Trace_PRO_CortexM_Tutorial](https://www.segger.com/downloads/jlink/J-Trace_PRO_CortexM_Tutorial.zip)

DerZc, 2019-12-23:
I'm so sorry that I have so much question.
I run the Strace example to get the trace, but the example do not return all the address in the trace.
the trace I want is
[image]
and the STrace example is:
[image]
the result I got is:
[image]
there are two address was missed: 0x80001b8 and0 x80001ba
I got other trace, sometime it is completed, but sometime some address was missed.
the project I run was come from the jlink website:
[J-Trace_PRO_CortexM_Tutorial](https://www.segger.com/downloads/jlink/J-Trace_PRO_CortexM_Tutorial.zip)

hkpeprah, 2019-12-23:
> Thank you for your help. May be the address I set was wrong, I reset the address and it run success.
> Now I can get the trace when it hit the breakpoint, but how can I get the instructions when it is running.

STrace only allows you to do a capture between two points; it stops at the second point (the break point), and you can dump the trace.  If you want a trace while the target is running, you have to use something like [ETM](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337h/CHDJAECG.html), but the FIFO queue can fill up, and wrap (I think), meaning you'll have to at some point stop the target, and flush the queue.  Reading the ETM registers may or may not be possible while the target is running; I've never tried that.

DerZc, 2019-12-23:
I know the ETM, I want to get the trace with C or python code, but I don't buy the JLink SDK. So I don't know which API can I use.

DerZc, 2019-12-29:
I meet a question that I can't start the RTT.
The board I use is STM32F407VE.
The main() is
[image]

And I get the result
[image]
How can I get the expection reason

DerZc, 2019-12-29:
I know the reason that I need restart after flash

DerZc, 2019-12-29:
I can run it, thank you very much

DerZc, 2020-01-03:
I meet some question in the rtt example.
My source code is that:
[image]
when I run the rtt example, I just can input only once, it while run automaticly with none input, the result is that:
[image]
the python file I run is the example you provide.

closed 2024-07-20 by hkpeprah