square/pylink #67: How to set or remove read protection of stm32

as564335sa, 2020-03-31:
Hi,
When I connect a chip that has read protection set, it will throw an exception，How can I set or remove read protection like jflash？

hkpeprah, 2020-04-14:
It should happen automatically if you've specified the device correctly.  There are unlock methods, but I've only implemented the unlock sequence for the Kinetis chip family.

Moral-ecomo, 2020-04-23:
Hi:
I just fix this problem, it just need a simple modify of the source code.
[](https://github.com/square/pylink/pull/70/files)

hkpeprah, 2020-04-23:
Whoops, you are correct, we by default specify `NO` to unlock automatically.

qtvbab, 2020-05-08:
HI,
I also encountered this problem。I used @Moral-ecomo  's method but still can't solve it。
How can I remove read protection of stm32？
The exception appears when connect():

   "D:\Program Files (x86)\Python38-32\lib\site-packages\pylink\jlink.py", line 1043, in connect
    result = self._dll.JLINKARM_Connect()
OSError: exception: access violation writing 0x00000001

qtvbab, 2020-05-12:
Hi, 
I successfully cleared the read protection with @Moral-ecomo method. However, a new problem has arisen. During continuous programming, Once in every two programming, jlink.reset() does not work, but no exception is thrown. Why this problem occurs，and how can I solve it

hkpeprah, 2020-05-12:
Which exception is thrown?  If you run with logging enabled, you should see more helpful messages:
```
import logging

logging.basicConfig(level=logging.DEBUG)
```

qtvbab, 2020-05-13:
Oh. I found the reason：
If the read protection is set while the debugger is still connected through JTAG/SWD, apply a
POR (power-on reset) instead of a system reset (without debugger connection).
So after the second programming, the chip cannot start after reset().