square/pylink #69: memory_read endianness issue

Sauci, 2020-04-11:
I'm facing to a strange behaviour regarding the ```memory_read32``` function. 

When I read data from a big endian target ([tms570lc4357zwt](https://www.ti.com/product/TMS570LC4357?utm_source=google&utm_medium=cpc&utm_campaign=epd-null-null-gpn_en-cpc-pf-google-eu&utm_content=tms570lc4357&ds_k=%257b_dssearchterm%257d&DCM=yes&gclid=CjwKCAjw1cX0BRBmEiwAy9tKHnrDxBQFeXmph9axT7CaFe5yc_5ppIsMtHVnhrL9rU1s9CQUDyLIexoCYawQAvD_BwE&gclsrc=aw.ds)), the resulting values returned from pylink are returned as big endian. However, doing the same operation in [Ozone](https://www.segger.com/products/development-tools/ozone-j-link-debugger/), the displayed values are correct.

As the variable I try to read holds the result of a function call, I also tried to read it using the ```register_read``` function, with **R0** as parameter. This function returns the expected value.

To summarize, knowing that the expected value is 6, and the 32-bits big endian representation of 6 is 100663296:
- Ozone shows **R0** = 6, as well as the value of my variable, **rst_source** = 6
- pylink returns 6 from ```register_read```, but the value of **rst_source** = 100663296 with ```memory_read32```

I'm not sure if the issue is in my script, the pylink package, or the dll itself, I just thought it was in the pylink package because of the mismatch between the value in **R0** and the value returned from ```memory_read32```

Thanks in advance for any feedacks

hkpeprah, 2020-04-16:
Sorry for getting back to this late.  Could you elaborate?  Say you have a 32-bit integer stored at memory address `0x12345678`, and you dereference and store said value in `R0` (explicitly), then if you pass that address to `memory_read32()`, it does not return `6`, whereas `register_read(r0)` returns `6`?

hkpeprah, 2020-04-16:
There are two methods: `set_little_endian()`, and `set_big_endian()`; could you try calling `set_little_endian()` before your read?

Sauci, 2020-04-16:
Thank you for your feedback.

You understood it correctly, here is the piece of assembly code (ARM none ABI):

```asm
...
bl      rst_get_reset_reason /* this function returns 6 */
ldr     r1, = rst_source /* rst_source variable is a 32 bit unsigned */
str     r0, [r1]
breakpoint: /* the breakpoint where the read is performed is located here */
...
```

The value is indeed correct when  calling ```set_little_endian``` before the ```memory_read32```. Otherwise, the function ```register_read('R0')``` returns 6, but ```memory_read32(<address of rst_source variable>)``` returns 100663296, which is 6 in opposite endianness.

For info, the version of the dll is 6.70a, and the connection is made with arguments  ```chip_name='tms570lc4357'```, which should set the endianness of the device correctly.

hkpeprah, 2020-04-16:
It's possible that it's something lower-level than this package than, unfortunately (possibly a regression).

Sauci, 2020-04-17:
Ok, thank you for your feedback. I opened a [ticket](https://forum.segger.com/index.php/Thread/7114-endianness-issue-on-tms570lc4357/) on the segger forum, I will give an update on this issue once I get a feedback from them.