square/pylink #76: Connect/Attach to existing session

stawiski, 2020-05-29:
Hello,

I'd like to attach to an existing session from pylink, instead of opening new connection the same way the JLink RTT Viewer does.
 
The `open()` and `connect()` APIs don't seem to have support for this.
I have also read through JLink docs and I can't find API to call as well.

[image]
[image]

Any ideas?

Cheers

stawiski, 2020-06-01:
@hkpeprah Thanks for your reply. Yeah i noticed there's no API for this in PyLink. I also couldn't find anything in SEGGER documentation. Too bad!

bernardfitch, 2020-08-24:
Hi @stawiski, I quite often want to call pylink API functions without first having to call `connect()` - I realise that this is only half of your requirement - but thought my solution might still help you.   

I've only had to make very minor changes to pylink in order to allow a 'connection' to be faked.   Essentially, all I've done is relax the test for a connection having been made in the `open_required()` and `connect_required()` decorators, e.g.:
```python
elif not self.connected():
```
allowing the tests to pass either if we truly are connected _or_ if a new 'connection faked' member variable is set, e.g.:
```python
elif not (self.connected() or self._connection_faked):
```
I intialise the new member variable to `False` in `__init__()` and have added a `fake_connection()` method to allow it to be set.  By default the module behaves as originally intended, however, if you call the new `fake_connection()` method the decorators allow you to call the decorated API calls even if no explicit call to `connect()` has been made.

I use this to allow me to implement the precise (bit-level) SWD connect sequence myself in situations where I want more control than is offered by the jlink DLL, but it can also be used in situations where another party/process has already initiated a connection.

I initially thought of altering the `connect()` or `connected()` methods, however, decided that leaving these methods true to the original concept of a connection makes the change both simpler to implement (as the other methods are called in many placed) and keeps the change in behaviour more transparent (as any 'faked' connection is explicit).

@hkpeprah - do you think that the changes I have described would be accepted if I made a pull request?  I would far rather be using a real version of plink than my hacked one and I'm sure other people would find the functionality useful.

Andrew ;)

PS. Apologies if I should really have started a new ticket - but I thought this was quite relevant to your question @stawiski

hkpeprah, 2020-08-24:
So we do have an `open()` and an `open_tunnel()`; the latter created to connect over the tunnel.  I would support adding a `connect` variant method (i.e. a new connect method) that connects to an already initiated connection provided we can implement some check to ensure it actually succeeds.  In that case, we can override the `.connected()` method to check that existing connection as well if a connection was made in that way.

hkpeprah, 2021-08-28:
@bernardfitch That sounds good to me.  We can continue the discussion in the PR.  I think my main concern is ensuring that the the connection type (fake vs. real) are abstracted as much as possible.

bernardfitch, 2021-08-28:
@hkpeprah Would you rather I submit two very small (but clearly targeted pull requests - one for the additional DLL API call wrapper and a second for the 'fake connection' changes) or a single pull request including both.  On reflection, two separate requests seems cleaner to me, but I'm happy to go with a single pull request if that would be less work for you.  If you have a strong preference then let me know, otherwise I'll probably go ahead with two (starting with the additional API function wrapper).

hkpeprah, 2021-08-30:
I think two PRs would be good, the SWD one should be fairly straightforward to merge without much discussion.