square/pylink #79: Unable to program NXP MIMXRT1064CVL5A using Pylink - supported devices look-up problem

rmadsen61, 2020-08-25:
In the jlink.py file (part of pylink) at approximately line #1055, there is a for loop used to search for the user-supplied chip name in a list of supported devices. I added a line within the for loop to write names of all of the supported devices to a disk file. There are several single entries in the resulting list that indicate support for a group devices, instead of just one device. Example: "MIMXRT1064xxx5A" appears in Segger's list of supported devices, one of only two MIMXRT1064-related list entries. Because pylink's search loop is performing a simple equality test to see if a specific chip name (such as "MIMXRT1064CVL5A") is in the list of supported devices, no specific chip name in the MIMXRT1064xxx5A group will ever be recognized as a supported device. "MIMXRT1064xxx5A" does not equal "MIMXRT1064CVL5A". Thus, setting the chip name to MIMXRT1064CVL5A and attempting to connect results in the exception message "'Unsupported device was connected to." To make things even more unhappy, if you set chip name to "MIMXRT1064xxx5A" to get pylink's search/equality test to work, then the device name "MIMXRT1064xxx5A" subsequently gets passed to Segger's JLink program, causing their software to fail during the attempt to program flash memory, because the passed-in chip name is not specific enough. If you look at all of the entries in Segger's list of supported devices, many of them are not simple device names that will properly match a specific chip name. This means that several devices will not be considered supported when they actually should be. The search for a supported device (chip name) should be made more sophisticated, to account for non-specific or otherwise complex entries in Segger's list. The specific chip name must then be passed to Segger's JLink software, as far as I can tell.

hkpeprah, 2020-08-27:
Ah, okay.  I think I get it.  Is your suggestion to treat the `x`s as wildcards, and then match them?  E.g.
```
>>> re.findall(re.sub(r'x', '.', 'MIMXRT1064xxx5A'), 'MIMXRT1064CVL5A')
['MIMXRT1064CVL5A']
```
I think we'd have to be careful to make sure that we're converting only multiple occurrences of lowercase `x`, but I don't see why not.  I'm not too much of a regex expert though to craft the appropriate substitution, but I welcome a patch if you want to take a stab at it.

rmadsen61, 2020-08-27:
Good afternoon. Attached is a document created with Libre Writer, detailing
my investigation into the problem I described on GitHub. I have also
attached a zipped text file that was generated by having the Python device
search loop print out all of the supported device names emanating from
Segger's list, so you can see the variation in device naming that can cause
a simple equality test to fail.

I will send you more information as I am able to discover, time permitting.

Thanks for your hard work to create pylink. It is deeply appreciated.

All the Best,
Ron Madsen

rmadsen61, 2020-08-27:
Just a brief addendum - you are right about making sure that conversion
only involves multiple occurrences of lower-case x. If Segger offers
support for a device that has an x in its actual part number, my simple
comparison function could detect a match when there really is not one.

Thanks again.

Ron Madsen