square/pylink #81: SWO output only works in debug mode?

Benjoyo, 2020-10-14:
I am trying out the Serial Wire Viewer example: [serial-wire-viewer](https://pylink.readthedocs.io/en/latest/swd.html#serial-wire-viewer) 

While it does work, it only seems to work when I put the device in debug mode from the SES IDE. 
Why is that? Can I do the same thing via pylink, so that the tool runs on its own?

Thanks

hkpeprah, 2020-10-14:
It could be possible that the IDE is writing to the registers necessary to configure SWO on your behalf.  The example above doesn't do that on your behalf.  An example of a firmware image that does do that is here: https://github.com/square/pylink/blob/master/tests/functional/firmware/k21-swo/src/swo.c#L75

While I haven't tried it myself, I think you should be able to write to those registers on the host-side, which would enable SWO then allow you to start collecting.

Benjoyo, 2020-10-18:
@hkpeprah Thank you for your reply. I tried the linked code to enable SWO, but it didn't change anything. It seems like the configuration doesn't have any effect, e.g. if I look at ITM_LAR before and after setting it, it is always zero. I also tried CMSIS, same thing. Is there anything else one needs to do first?
I am using an nrf52840 DK, which also has a Cortex M4.

hkpeprah, 2020-10-20:
Could this be of use? https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/debugging-nrf52-series-devices-using-trace

boraozgen, 2021-09-23:
@Benjoyo were you able to solve this? 

I am also struggling to get SWO working. It works if I run the JLinkSWOViewer tool before the example, otherwise it does not work. I wonder what the tool might be doing additionally.

closed 2025-01-28 by hkpeprah