square/pylink #83: Can we replace the pusutil.pid_exists with another method?

Hoohaha, 2020-11-02:
Hi,

pylink is a greate tool and we use it to test MCU devices.
However I always met some trobules if install it on windows 10 system.
It fails to when installing psutil module and it report missing Visual Studio C++2015...
I have to cost many time to download vistual studio installer and then install C++ build tools. 

I checked the code and there only one place to use psutil.pid_exists check pid is alive.
Can we remove the module psutil and replace it with other workaround?
https://stackoverflow.com/questions/568271/how-to-check-if-there-exists-a-process-with-a-given-pid-in-python

hkpeprah, 2020-11-02:
I believe there was an open pull request in the same vein (#68); my concern with what is proposed in the StackOverflow thread is that it doesn't cover all operating systems; taking a look at `psutil`, they have multiple different implementations of `pid_exists()` depending on the flavour of *nix, the flavour of Windows, or if you're on OSX.  However, since this has been asked more than once, I think adding a install option that allows one to use it without `psutil` might make sense; something like an [`install option`](https://pip.pypa.io/en/stable/reference/pip_install/#cmdoption-install-option); this would require updating the `setup.py`, and adding in the `pid_exists()` code (probably from #68) with a `try-catch` of sorts.  E.g.

```
# util.py
def _pid_exists():
   # internal implementation

def pid_exists(pid):
    try:
        import psutil
        return psutil.pid_exists(pid)
    except ImportError:
        return _pid_exists(pid)
```

Hoohaha, 2020-11-04:
I think it is acceptable, this makes pylink easy to install.

FARLY7, 2022-01-10:
Hi there @hkpeprah was there any progress on this?

I used the `pylink` library a lot in our internal and external tooling, but since it requires our Windows users to perform a huge install of Visual Studio C++ IDE and its build tools etc, I have unfortunately had to drop it and opt for commander scripts instead.

This would be a great feature if added. It seems 'broken' to have to install such large prerequisites.

hkpeprah, 2022-01-11:
Hi @FARLY7, sorry about that. I currently don't have the bandwidth to work on this, but would gladly accept a modification of #68 that doesn't completely drop `psutil`, but instead allows for the package to be installed without `psutil` via some sort of install command option.