square/pylink #87: Incorrect decorators in raw pin operation functions in the JLink class

sgmne, 2021-02-28:
The functions in JLink class

set_reset_pin_high(self):
set_reset_pin_low(self):
set_tck_pin_high(self):
set_tck_pin_low(self):
set_tdi_pin_high(self):
set_tdi_pin_low(self):
set_tms_pin_high(self):
set_tms_pin_low(self):
set_trst_pin_high(self):
set_trst_pin_low(self):

are raw operations and hence should be @open_required, not @connection_required.

hkpeprah, 2021-03-01:
Thanks for bringing this up, this is definitely an oversight.  If you wish to submit a patch to address this, I can review and merge it in, otherwise I will open up one when I find time available.

sgmne, 2021-03-01:
[0001-Change-raw-pin-set-high-low-function-decorators-from.zip](https://github.com/square/pylink/files/6063286/0001-Change-raw-pin-set-high-low-function-decorators-from.zip)
Patch enclosed.