square/pylink #91: Setting SWO speed above 50 MHz on J-Trace

graeme-hattan, 2021-04-01:
According to Segger, the J-Trace PRO models are capable of SWO speeds up to [100 MHz](https://www.segger.com/products/debug-probes/j-trace/models/model-overview/).

However, when I call jlink.swo_start(100_000_000), I the following dialog pops up:

[image]

Is there a ways to enable speeds above 50 MHz?

hkpeprah, 2021-04-01:
Are you sure you have the right device plugged in?  The error dialog suggests you're selecting `100000kHz` as the speed, but your device only supports a maximum speed of `50000kHz`.

graeme-hattan, 2021-04-01:
Yes, definitely, I only have one Segger device and it says J-Trace Pro on it!