square/pylink #92: Detect connected debugger

Viatorus, 2021-04-09:
Is it possible to get a list of all connected debuggers so one could select one of those instead of providing a serial number?

hkpeprah, 2021-04-09:
Yup.  You can do so using the [`connected_emulators()`](https://pylink.readthedocs.io/en/latest/pylink.bindings.html#pylink.structs.JLinkConnectInfo) method, which returns a list of connection information devices, see the [docs here](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.connected_emulators).  You can then use the returned list to figure out the serial number of IP address to pass to your call to `open()`.

Viatorus, 2021-04-12:
Great, thank you! I was confused by the name emulator.

closed 2021-04-12 by Viatorus