square/pylink #94: connect() doesn't work - need to connect using the JFlash(Exe) UI to successfully use pylink

kettenbach-it, 2021-04-09:
Hi,

on MacOS and Linux with Segger Tools V6.98A I need to open the GUI and connect to the target to make my program work.
"Only" calling pylink.connect() doesn't connect.

Code:
```
jlink = pylink.JLink()
print("JLink opened: " + str(jlink.opened()))
print("JLink connected: " + str(jlink.connected()))
print("JLink productname: " + jlink.product_name)
jlink.connect(chip_name="nRF52832_xxAA", verbose=True)
print(("JLink target connected: " + str(jlink.target_connected())))
if jlink.target_connected():
    print("Device Familiy: " + str(jlink.device_family()))
    print("Core ID: " + str(jlink.core_id()))
    print("Core CPU: " + str(jlink.core_cpu()))
.... do some more things like flashing
```

Without using the UI to connect:
```
JLink opened: True
JLink connected: True
JLink productname: J-Link OB-SAM3U128-V2-NordicSemi compiled Feb  2 2021 16:47:20
JLink target connected: False
```

Running the UI and clicking on "connect":

```
JLink opened: True
JLink connected: True
JLink productname: J-Link OB-SAM3U128-V2-NordicSemi compiled Feb  2 2021 16:47:20
JLink target connected: True
Device Familiy: 14
Core ID: 731911287
Core CPU: 234881279
etc.
```

hkpeprah, 2021-04-09:
You're missing a call to `.open()` before `.connect()`.  Without the UI, you will need to use `.open()` to select which emulator you want to use (as you can have multiple connected to your machine at a given time), then `.connect()` to connect through the emulator to the target.  `.open()` takes either a serial number or IP address; you can use `.connected_emulators()` before `.open()` if you have multiple emulators in order to determine which one to use.

kettenbach-it, 2021-04-09:
Ah, sorry. Mac code example wasn't complete. I missed the "open" line.
Please see above

hkpeprah, 2021-04-09:
Is it possible that you need to set the target interface via [`.set_tif()`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.set_tif)?  By default, the interface is JTAG with default values for the JTAG configuration ([`jtag_configure()`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.jtag_configure) is used to change these.  For example, if the interface is SWD, you would call:
```
jlink.open()
jlink.set_tif(enums.JLinkInterfaces.SWD)
jlink.connect('....')
```

kettenbach-it, 2021-04-09:
Perfect! That fixed it! Thank you **SOO** much!

hkpeprah, 2021-04-09:
Happy to help.

closed 2021-04-09 by hkpeprah