square/pylink #98: UniCodeDecodeError(s) seen with pylink 0.10.0 on Ubuntu 18.04.5 LTS

karthik0134, 2021-04-26:
Environment : 
------------------------------
Ubuntu 18.04.5 LTS (on a HP Prodesk https://support.hp.com/us-en/document/c05364047)
CPU: Intel Core i5-7500T@2.70 GHz x4 (4-Core) - systems that show the problem.

Memory: 8 GB
Disk : 250 GB
OS: 64-bit, Ubuntu 18.04.5 LTS
 Release:	18.04
 Codename:	bionic
--
JLink hardware used: JLink base,   https://shop-us.segger.com/J_Link_p/8.08.00.htm with FW version 6.80

Expected vs actual behavior:
------------------------------
Existing python code that works well with pylink-square 0.9.0 gives UnicodeDecodeErrors with pylink-square 0.10.10
and the above JLink configuration. Error messages are shown below. It appears it only happens on certain Ubuntu 18.04.5 LTS systems (have 3 always repro'd hosts but not 10 others). It might be that pylink-square is dependent on some other python packages in the system to show this error, otherwise it would show up everywhere. For what it's worth, the other 10+ Ubuntu systems have more powerful config, but I don't think that's the issue since the SW is the same.

CPU: Intel Core i5-9500T@2.20 GHz x6 (6-Core) - systems that don't show the problem (pylink-square 0.10.0 works fine)
Memory: 16 GB
Disk : 483 GB
OS: 64-bit, Ubuntu 18.04.5 LTS

Error messages:
------------------------------
"Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 323, in <lambda>
    self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x81 in position 27: invalid start byte
  .....

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 323, in <lambda>
    self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 28: invalid continuation byte
  ....

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 323, in <lambda>
    self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8f in position 27: invalid start byte
  ....

Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 323, in <lambda>
    self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x97 in position 26: invalid start byte
  ....
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 232, in 'calling callback function'
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 323, in <lambda>
    self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 25: invalid continuation byte"
....

karthik0134, 2021-04-26:
The above error messages are seen when connecting to a JLink with a specified serial # 60126753, see below.

  " self.detailed_log_handler = lambda s: (detailed_log or logger.debug)(s.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 25: invalid continuation byte
   INFO 13:49:27 [<filename>.py:  98] Connecting to device
	Chip: nRF52832_xxAA
	Serial:50126753
...
Traceback (most recent call last):
  ... 5 other source files that finally boil down to...
  File "/home/test/workdir/12056/workspace/.env/lib/python3.7/site-packages/pylink/jlink.py", line 723, in open
    raise errors.JLinkException('No emulator with serial number %s found.' % serial_no)
pylink.errors.JLinkException: No emulator with serial number 50126753 found."

scottard, 2021-07-05:
I see the same error on rasperry pi 4 model B with:
Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux

I connected the jlink to a NRF52 chip.

I tried with JLink_Linux_V672d_arm.tgz and the error is not present but as soon as I try to use the version 7 (I tried with JLink_Linux_V700a_arm.deb and JLink_Linux_V750_arm.deb) the problem appears.

scottard, 2021-07-05:
Tried with JLink_Linux_V698e_arm.deb and the error is present so it is not related to the version 7