square/pylink #102: module name conflict

mtrzas, 2021-06-14:
I have a conflict when pylink(https://pypi.org/project/PyLink/) and pylink-square are installed on my device.
Both modules are imported with :

`import pylink`

When pylink-square is installed before pylink then I get this error:

`AttributeError. module 'pylink' has no attribute 'JLink'`

Is it possible to rename the module?

hkpeprah, 2021-06-16:
This is definitely an issue.  Unfortunately, I think renaming the module would have wide reaching consequences for downstream consumers.  If you're not already using it, would maybe a [virtual environment](https://docs.python.org/3/tutorial/venv.html) help with your development flow?

mtrzas, 2021-06-18:
Thank you for your Answer :)

I already using a virtual environment.
For me it is not a big problem because I'm not using pylink(https://pypi.org/project/PyLink/) and I'm only using pylink-square.
I find this issue on accident.

onurt91, 2023-08-18:
Hello, how did you manage to solve the issue ?
I can have both pylink  or pylink-square (together or only one at a time) and still have the problem : 
"AttributeError. module 'pylink' has no attribute 'JLink'"

Thank you