square/pylink #108: [bug] Unsupported device selected

Hoohaha, 2021-08-31:
**General Description**

We added custom device support in J-Link, by adding device info in JLinkDevices.xml, and provide flash algo in Devices directory.
It works with JLink.exe, but failed with pylink, the pylink always report Unsupported device selected.

Checked code, pylink always copy DLL to a temp folder. I commented copy related code, and use origin DLL file, then the issue is gone. So I guess the DLL will look up the JLinkDevices.xml if it not found in DLL internal.

** Version **
JLink: 7.50
Python 3.8

hkpeprah, 2021-09-03:
Sorry about the late response here.  I think this has been raised a few times before.  A workaround that some have done is sub-classing the `Library` class, and implementing their own `load()` method (so that it doesn't copy the DLL), but now that I think about it some more, I think there are possibly two things we could do to make this better:

1. When copying the library to a temporary folder, also copy any found XML files (this would address the issue where users have a custom devices XML).
2. Provide a sub-class of the `Library` instance that doesn't do the copying, and allow users to pass this in themselves when creating their emulator instance.

I think both of these would address the problem with a solution that works for most users.  I don't have the bandwidth at the moment to do both, but could possibly in the future, but if you (or anyone else) is interested in tackling it, I would gladly review and accept the changes.

The behaviour around the DLL copying is to resolve an issue in earlier DLL versions where it wasn't possible to debug multiple chips using multiple emulators at the same time (locking issue).  The copying works around that so that someone can debug multiple devices at the same time.

Hoohaha, 2022-07-22:
#32 

Use this workaround can resolve this issue

```python
  xml_path = Path(jlink._library._path).parent / "JLinkDevices.xml"
  jlink.exec_command(f'JLinkDevicesXMLPath {xml_path.as_posix()}')
```