square/pylink #115: Running PyLink on Apple M1 macs with x86_64 Python

ccattuto, 2021-11-27:
When using PyLink on Apple M1 machines with an x86_64 version of Python, OSX uses Rosetta to translate the x86 Python binaries to arm64 instructions. However, this means that the x86_64 Python programs run under Rosetta are constrained to _dynamically load x86_64 libraries only, and cannot load arm64 dynamic libraries_. This creates an issue when PyLink attempts to load the J-Link dynamic libraries, as ctype finds the arm64, and fails to load them as follows:
  `/var/folders/qs/xdp9zrt56wj0vcw28_7b_7w00000gn/T/tmp7xiecla1.dylib: mach-o, but wrong architecture`

Possible solutions are:

- Recommending users to only run PyLink (and Python software depending on it, e.g., PyOCD) using native (arm64) versions of Python. This is currently tricky as, e.g., Anaconda still hasn't shipped an arm64 native Python distribution.
- Recommending PyLink users to install the x86_64 version of J-Link rather than the native arm64 one.
- Modify PyLink to recognize that it is running under Rosetta and choose accordingly the x86_64 dynamic J-Link library rather than the arm64 one (recommending users to install both versions).

I have temporarily solved the issue by installing both the arm64 and x86_64 versions of J-Link, making sure that the latter is installed in a directory whose name contains `x86_64`, and modifying `library.py` to force loading the x86_64 library as follows:
```
elif sys.platform.startswith('darwin'):
    #path = next(self.find_library_darwin(), None)
    path = next(filter(lambda s: 'x86_64' in s, self.find_library_darwin()), None)
```

hkpeprah, 2021-12-12:
Sorry about the late response here, things have been a bit busy with work and a new dog. I think the idea of modifying the library to determine if it is running under Rosetta and choose the library appropriately makes sense. I don't have an M1 myself, but we have received one or two on the team as exploratory dev devices, so the work will eventually get done. However, if you're interested in submitting a patch, I'm more than happy to work with a colleague to review and approve.

ccattuto, 2021-12-12:
Thanks! Unfortunately, I don't have the bandwidth to work on this myself, but I just wanted to make sure that the issue is documented and understood, in case other users encounters it.