square/pylink #116: Instructions on dealing with external dependencies on Mac

ccattuto, 2021-11-27:
On recent versions of OSX (tested on Big Sur), it is by default impossible to set some environment variables, including `DYLD_LIBRARY_PATH`. Hence the suggested Option B, reported below, does not work.
```
# Option B: Add SEGGER's J-Link directory to your dynamic libraries path.
$ export DYLD_LIBRARY_PATH=/Applications/SEGGER/JLink:$DYLD_LIBRARY_PATH
```

hkpeprah, 2021-12-12:
Sorry about the long response here. I am on BIG Sur and I am able to export the path. What are you seeing get output when you do?

ccattuto, 2021-12-12:
`DYLD_LIBRARY_PATH` can be set, but System Integrity Protection on OSX will "sanitize" it.
Example:
```
% export DYLD_LIBRARY_PATH=/Applications/SEGGER/JLink:$DYLD_LIBRARY_PATH
% echo $DYLD_LIBRARY_PATH
/Applications/SEGGER/JLink:

% cat test.sh
#!/bin/sh
echo $DYLD_LIBRARY_PATH

% ./test.sh

%
```
See also this: https://en.wikipedia.org/wiki/System_Integrity_Protection
>System Integrity Protection will also sanitize certain environmental variables when calling system programs when SIP is in effect. For example, SIP will sanitize LD_LIBRARY_PATH and DYLD_LIBRARY_PATH before calling a system program like /bin/bash to avoid code injections into the Bash process. [..]