square/pylink #119: No sdist on pypi.org

dvzrv, 2022-01-20:
Hi! For packaging purposes downstreams such as Arch Linux rely on sdist tarballs, such as those on pypi.org. Currently you only have .whl files there.

Do you have plans of adding sdist tarballs as well? :)

hkpeprah, 2022-01-21:
No current plans, but this seems fairly straightforward. We used to release tarballs, but then switched to wheels. I think it's possible to generate both though, though it's been awhile since I looked at the Python packaging documentation. I would gladly accept a patch for this, barring that, I should be able to knock something out myself within a week or so if it's just a straightforward configuration option.

dvzrv, 2022-04-04:
@hkpeprah thanks for looking into that! The pypi sdist tarball for 0.12.0 unfortunately lacks the tests though, so downstreams will not be able to run them for this project when building a package.

As [MANIFEST.in](https://github.com/square/pylink/blob/31284d29da5d6501bb247e6d9678806761404ee3/MANIFEST.in) is still in use, it could be used to add the tests directory to the sdist tarball for future releases.

hkpeprah, 2022-05-26:
Should be fixed in `0.13.0`: https://github.com/square/pylink/releases/tag/v0.13.0

dvzrv, 2022-06-07:
Unfortunately that is still an issue with the sdist tarball on pypi.org:

```
drwxr-xr-x runner/docker     0 2022-05-30 22:38 pylink-square-0.13.0/
-rw-r--r-- runner/docker 12884 2022-05-30 22:38 pylink-square-0.13.0/CHANGELOG.md
-rw-r--r-- runner/docker 10181 2022-05-30 22:38 pylink-square-0.13.0/LICENSE.md
-rw-r--r-- runner/docker   104 2022-05-30 22:38 pylink-square-0.13.0/MANIFEST.in
-rw-r--r-- runner/docker  5035 2022-05-30 22:38 pylink-square-0.13.0/PKG-INFO
-rw-r--r-- runner/docker  4439 2022-05-30 22:38 pylink-square-0.13.0/README.md
drwxr-xr-x runner/docker     0 2022-05-30 22:38 pylink-square-0.13.0/examples/
-rwxr-xr-x runner/docker  5562 2022-05-30 22:38 pylink-square-0.13.0/examples/pylink-rtt
-rwxr-xr-x runner/docker  3202 2022-05-30 22:38 pylink-square-0.13.0/examples/pylink-swv
drwxr-xr-x runner/docker     0 2022-05-30 22:38 pylink-square-0.13.0/pylink/
-rw-r--r-- runner/docker  1144 2022-05-30 22:38 pylink-square-0.13.0/pylink/__init__.py
-rw-r--r-- runner/docker 20657 2022-05-30 22:38 pylink-square-0.13.0/pylink/__main__.py
-rw-r--r-- runner/docker  2138 2022-05-30 22:38 pylink-square-0.13.0/pylink/binpacker.py
-rw-r--r-- runner/docker  3080 2022-05-30 22:38 pylink-square-0.13.0/pylink/decorators.py
-rw-r--r-- runner/docker 21561 2022-05-30 22:38 pylink-square-0.13.0/pylink/enums.py
-rw-r--r-- runner/docker  1970 2022-05-30 22:38 pylink-square-0.13.0/pylink/errors.py
-rw-r--r-- runner/docker 167995 2022-05-30 22:38 pylink-square-0.13.0/pylink/jlink.py
-rw-r--r-- runner/docker   4881 2022-05-30 22:38 pylink-square-0.13.0/pylink/jlock.py
-rw-r--r-- runner/docker  15099 2022-05-30 22:38 pylink-square-0.13.0/pylink/library.py
drwxr-xr-x runner/docker      0 2022-05-30 22:38 pylink-square-0.13.0/pylink/protocols/
-rw-r--r-- runner/docker    576 2022-05-30 22:38 pylink-square-0.13.0/pylink/protocols/__init__.py
-rw-r--r-- runner/docker   9106 2022-05-30 22:38 pylink-square-0.13.0/pylink/protocols/swd.py
-rw-r--r-- runner/docker  10368 2022-05-30 22:38 pylink-square-0.13.0/pylink/registers.py
-rw-r--r-- runner/docker  38009 2022-05-30 22:38 pylink-square-0.13.0/pylink/structs.py
-rw-r--r-- runner/docker   2185 2022-05-30 22:38 pylink-square-0.13.0/pylink/threads.py
drwxr-xr-x runner/docker      0 2022-05-30 22:38 pylink-square-0.13.0/pylink/unlockers/
-rw-r--r-- runner/docker   1155 2022-05-30 22:38 pylink-square-0.13.0/pylink/unlockers/__init__.py
-rw-r--r-- runner/docker   9031 2022-05-30 22:38 pylink-square-0.13.0/pylink/unlockers/unlock_kinetis.py
-rw-r--r-- runner/docker   4598 2022-05-30 22:38 pylink-square-0.13.0/pylink/util.py
drwxr-xr-x runner/docker      0 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/
-rw-r--r-- runner/docker   5035 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/PKG-INFO
-rw-r--r-- runner/docker    678 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/SOURCES.txt
-rw-r--r-- runner/docker      1 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/dependency_links.txt
-rw-r--r-- runner/docker     49 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/entry_points.txt
-rw-r--r-- runner/docker     25 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/requires.txt
-rw-r--r-- runner/docker      7 2022-05-30 22:38 pylink-square-0.13.0/pylink_square.egg-info/top_level.txt
-rw-r--r-- runner/docker    199 2022-05-30 22:38 pylink-square-0.13.0/setup.cfg
-rw-r--r-- runner/docker   7429 2022-05-30 22:38 pylink-square-0.13.0/setup.py
```

hkpeprah, 2022-07-20:
I'm still looking at this in between other work, though I'm not sure what is going on since I see the tarball in my build. Will have to do more research to see if it's getting stripped by PyPi for some reason.

dvzrv, 2022-07-21:
> As [MANIFEST.in](https://github.com/square/pylink/blob/31284d29da5d6501bb247e6d9678806761404ee3/MANIFEST.in) is still in use, it could be used to add the tests directory to the sdist tarball for future releases.

As mentioned earlier, you'll have to adapt MANIFEST.in to include the tests in the sdist tarball. See https://setuptools.pypa.io/en/latest/deprecated/distutils/sourcedist.html#manifest for further information.

hkpeprah, 2022-07-21:
Hm, I thought I did add them here: https://github.com/square/pylink/blob/master/MANIFEST.in

hkpeprah, 2022-07-21:
Oof, I messed up and did `recursive-exclude`; I meant `recursive-include` for the `tests` 🤦 
Will fix.

dvzrv, 2022-08-13:
Still none in 0.14.2:

```
drwxr-xr-x runner/docker     0 2022-08-12 23:12 pylink-square-0.14.2/
-rw-r--r-- runner/docker 13303 2022-08-12 23:11 pylink-square-0.14.2/CHANGELOG.md
-rw-r--r-- runner/docker 10181 2022-08-12 23:11 pylink-square-0.14.2/LICENSE.md
-rw-r--r-- runner/docker   122 2022-08-12 23:11 pylink-square-0.14.2/MANIFEST.in
-rw-r--r-- runner/docker  5015 2022-08-12 23:12 pylink-square-0.14.2/PKG-INFO
-rw-r--r-- runner/docker  4439 2022-08-12 23:11 pylink-square-0.14.2/README.md
drwxr-xr-x runner/docker     0 2022-08-12 23:12 pylink-square-0.14.2/examples/
-rwxr-xr-x runner/docker  5562 2022-08-12 23:11 pylink-square-0.14.2/examples/pylink-rtt
-rwxr-xr-x runner/docker  3202 2022-08-12 23:11 pylink-square-0.14.2/examples/pylink-swv
drwxr-xr-x runner/docker     0 2022-08-12 23:12 pylink-square-0.14.2/pylink/
-rw-r--r-- runner/docker  1144 2022-08-12 23:11 pylink-square-0.14.2/pylink/__init__.py
-rw-r--r-- runner/docker 20657 2022-08-12 23:11 pylink-square-0.14.2/pylink/__main__.py
-rw-r--r-- runner/docker  2138 2022-08-12 23:11 pylink-square-0.14.2/pylink/binpacker.py
-rw-r--r-- runner/docker  3080 2022-08-12 23:11 pylink-square-0.14.2/pylink/decorators.py
-rw-r--r-- runner/docker 21561 2022-08-12 23:11 pylink-square-0.14.2/pylink/enums.py
-rw-r--r-- runner/docker  1970 2022-08-12 23:11 pylink-square-0.14.2/pylink/errors.py
-rw-r--r-- runner/docker 167995 2022-08-12 23:11 pylink-square-0.14.2/pylink/jlink.py
-rw-r--r-- runner/docker   4881 2022-08-12 23:11 pylink-square-0.14.2/pylink/jlock.py
-rw-r--r-- runner/docker  21538 2022-08-12 23:11 pylink-square-0.14.2/pylink/library.py
drwxr-xr-x runner/docker      0 2022-08-12 23:12 pylink-square-0.14.2/pylink/protocols/
-rw-r--r-- runner/docker    576 2022-08-12 23:11 pylink-square-0.14.2/pylink/protocols/__init__.py
-rw-r--r-- runner/docker   9106 2022-08-12 23:11 pylink-square-0.14.2/pylink/protocols/swd.py
-rw-r--r-- runner/docker  10368 2022-08-12 23:11 pylink-square-0.14.2/pylink/registers.py
-rw-r--r-- runner/docker  38009 2022-08-12 23:11 pylink-square-0.14.2/pylink/structs.py
-rw-r--r-- runner/docker   2185 2022-08-12 23:11 pylink-square-0.14.2/pylink/threads.py
drwxr-xr-x runner/docker      0 2022-08-12 23:12 pylink-square-0.14.2/pylink/unlockers/
-rw-r--r-- runner/docker   1155 2022-08-12 23:11 pylink-square-0.14.2/pylink/unlockers/__init__.py
-rw-r--r-- runner/docker   9031 2022-08-12 23:11 pylink-square-0.14.2/pylink/unlockers/unlock_kinetis.py
-rw-r--r-- runner/docker   4598 2022-08-12 23:11 pylink-square-0.14.2/pylink/util.py
drwxr-xr-x runner/docker      0 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/
-rw-r--r-- runner/docker   5015 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/PKG-INFO
-rw-r--r-- runner/docker    678 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/SOURCES.txt
-rw-r--r-- runner/docker      1 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/dependency_links.txt
-rw-r--r-- runner/docker     48 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/entry_points.txt
-rw-r--r-- runner/docker     25 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/requires.txt
-rw-r--r-- runner/docker      7 2022-08-12 23:12 pylink-square-0.14.2/pylink_square.egg-info/top_level.txt
-rw-r--r-- runner/docker    199 2022-08-12 23:12 pylink-square-0.14.2/setup.cfg
-rw-r--r-- runner/docker   7429 2022-08-12 23:11 pylink-square-0.14.2/setup.py
```

You can try it before release by creating a tarball locally and looking into it: ` tar -tvf <filename>`

hkpeprah, 2022-10-03:
It took me awhile to figure this out (and was delayed by other work), but I think `0.14.3` should finally have the tests available.