square/pylink #123: Write to mcu register

witc, 2022-02-25:
Hi,

this pylink is realy great tool! Everything work perfectly. But:
Now I am trying to write to GPIO register (just write 1 to one bit), which turns LED ON on the GPIO pin. I am not sure how to do that. I tried `memory_write(int(0x50000c18),[int(0b100)]) , also flash_write(int(addrSWIER),[int(0b100)],32) ` but nothing works. When I want to write data to memory it works perfect, but these maybe is different area and it does not work.  (I Use STM32L)

Can you help me please?

Thank you very much. Jan.

witc, 2022-02-26:
Resolved!

Just added halt() before memory_write(...) and after that I add function mcuRun (which calls _dll.JLINKARM_Go())

closed 2022-02-26 by witc

hkpeprah, 2022-03-02:
Sorry about getting back to this late. Yup, you need to stop the core to write to memory.