square/pylink #130: Erase range?

mburr, 2022-05-16:
Segger's J-Link Commander utility allows an address range to be passed to the erase command.  I don't see that capability in pylink's `erase()` function which calls the `JLINK_EraseChip()` API.

Is there a way to erase a range of flash that I'm not seeing?

hkpeprah, 2022-05-16:
I don't believe so. Is it possible that calling `flash()` with a buffer full of `1`s would give the erase behaviour? I've not tried it myself. E.g.
```
programemr = pylink.JLink(..)
addr = 0xDEADBEEF
region_size = 0xFFFF
data = [1] * region_size
programmer.flash_write(data, addr)
```

mburr, 2022-05-16:
Yes, that should work.  I should have thought of that myself!  Thanks.

closed 2022-05-16 by mburr

denravonska, 2022-11-23:
Did it work though? It's usually only possible to flip a bit from `1` to `0` with the other way around requiring an erase. Does the JLink erase sectors for you behind the scenes?

mburr, 2022-11-23:
Sorry - I never actually got around to trying it, because I was able to complete my work using Segger's Jlink Commander tool so I didn't need to integrate pylink into my python script.