square/pylink #134: compatible_firmware_version is broken with JLink V7.58b

juhhov, 2022-06-09:
```
$ jlink.compatible_firmware_version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/pylink/jlink.py", line 142, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pylink/jlink.py", line 1232, in compatible_firmware_version
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.
```

Call works with JLink V6.88a.

juhhov, 2022-06-10:
The last version where compatible_firmware_version works is V7.52a.

hkpeprah, 2022-06-13:
Sorry about the late response here. Are there any logs if you run with verbosity enabled?

juhhov, 2022-06-14:
Sure, here you go:
```
$ pylink -vvv firmware -u -s 821005721
DEBUG:pylink.jlink:T1BECC740 000:172.684   
DEBUG:pylink.jlink:Firmware: J-Link V11 compiled JUN 17 2021 16:42:11
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:173.140   
DEBUG:pylink.jlink:Firmware: J-Link V11 compiled JUN 17 2021 16:42:11
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:192.145   
DEBUG:pylink.jlink:Hardware: V11.00
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:192.299   
DEBUG:pylink.jlink:S/N: 821005721
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:192.365   
DEBUG:pylink.jlink:OEM: SEGGER
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:192.431   
DEBUG:pylink.jlink:Feature(s): GDB
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:194.001   
DEBUG:pylink.jlink:TELNET listener socket opened on port 19021
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:194.142   
DEBUG:pylink.jlink:WEBSRV WEBSRV_Init(): Starting webserver thread(s)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:194.254   
DEBUG:pylink.jlink:WEBSRV Webserver running on local port 19080
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:194.332 
DEBUG:pylink.jlink:- 22.924ms returns "O.K."

DEBUG:pylink.jlink:T1BECC740 000:194.424 
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog

DEBUG:pylink.jlink:T1BECC740 000:194.476   
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog(...)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:T1BECC740 000:194.542 
DEBUG:pylink.jlink:- 0.140ms returns 0

DEBUG:pylink.jlink:T1BECC740 000:194.596 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:194.640 
DEBUG:pylink.jlink:- 0.065ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:194.687 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:194.730 
DEBUG:pylink.jlink:- 0.064ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:194.780 
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T1BECC740 000:194.824 
DEBUG:pylink.jlink:- 0.065ms returns TRUE

DEBUG:pylink.jlink:T1BECC740 000:194.873 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:194.917 
DEBUG:pylink.jlink:- 0.064ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:194.960 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:195.003 
DEBUG:pylink.jlink:- 0.064ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:195.046 
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T1BECC740 000:195.089 
DEBUG:pylink.jlink:- 0.063ms returns TRUE

DEBUG:pylink.jlink:T1BECC740 000:195.136 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:195.180 
DEBUG:pylink.jlink:- 0.065ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:195.224 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:195.266 
DEBUG:pylink.jlink:- 0.063ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:195.311 
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T1BECC740 000:195.353 
DEBUG:pylink.jlink:- 0.064ms returns TRUE

DEBUG:pylink.jlink:T1BECC740 000:195.444 
DEBUG:pylink.jlink:JLINK_GetFirmwareString(...)

DEBUG:pylink.jlink:T1BECC740 000:195.495 
DEBUG:pylink.jlink:- 0.073ms

DEBUG:pylink.jlink:T1BECC740 000:195.553 
DEBUG:pylink.jlink:JLINK_GetEmbeddedFWString(...)

DEBUG:pylink.jlink:T1BECC740 000:210.915 
DEBUG:pylink.jlink:- 15.442ms

Error: Unspecified error.
DEBUG:pylink.jlink:T1BECC740 000:211.077 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:211.132 
DEBUG:pylink.jlink:- 0.079ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:211.181 
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:T1BECC740 000:211.228 
DEBUG:pylink.jlink:- 0.069ms returns TRUE

DEBUG:pylink.jlink:T1BECC740 000:211.277 
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:T1BECC740 000:211.323 
DEBUG:pylink.jlink:- 0.068ms returns 0x01

DEBUG:pylink.jlink:T1BECC740 000:214.433 
DEBUG:pylink.jlink:JLINK_Close()
```

hkpeprah, 2022-07-20:
It might be an issue in the shared library unfortunately, which we can't fix in this Python library, unfortunately :/

bbrown1867, 2023-01-11:
I saw this issue as well with a J-Link Base Compact + software v7.58b (universal macOS). I upgraded to v7.84b and it still persisted. Only workaround was manually stub `firmware_outdated` and `firmware_newer` to return `False`. I believe it is something to do with `pylink`, since using `JLinkExe` works fine and a firmware updated occurred the first time I used it.

I wonder if something has changed with the call to `JLINKARM_GetEmbeddedFWString`, as it seems to be returning -1 every time.

bbrown1867, 2025-02-19:
This issue is resolved by #223.

In older versions of `pylink`, it can be resolved by setting `use_tmpcpy=False`.

closed 2025-02-19 by bbrown1867