square/pylink #135: erase() and flash_file functions are not working

Dave5217, 2022-06-17:
I am trying to automate the erasing and reflashing of a chip but none of them are working.
Here is my code:

import pylink
import time

error = ''
file = 'example.bin'
jlink = pylink.JLink()

jlink.open(serial_no=601011318)
jlink.set_tif(1)

try:
    jlink.connect('CORTEX-M0', 4000)
    jlink.erase()
    sleep(5)
    jlink.flash_file(file,0x0)
    
except error:
    print('Couldn't do it')
    
finally:
    jlink.close()

The erase function doesn't erase the chip but throws no exception. The flash_flie function throws the following exception:
JLinkFlashException: Could not write target memory.

hkpeprah, 2022-06-17:
You likely need to stop the core before flashing it using: `.halt()`.

Dave5217, 2022-06-17:
I saw that the method checks if the core is halted and halts it if it's not, but anyway I tried to stop the core first and got the same exception

hkpeprah, 2022-06-20:
Hm, usually an error like that happens if the device isn't stopped, or if it has to be reset in a special mode (see [here](https://pylink.readthedocs.io/en/latest/troubleshooting.html#unspecified-error-during-erase) for troubleshooting ideas). If you run with verbosity enabled, you should be able to see more logs that should point to where the issue is (e.g. "Failed to Stop Core").