square/pylink #136: Changing a specific memory address

GabrielMarconS, 2022-06-30:
Hey, I'm new to Python and pylink, but I've managed to flash my target through file. I wanted to know if is it possible to change a specific memory address, if so, any tips?

hkpeprah, 2022-06-30:
Hello,

Do you mean changing the last few bytes of the image you're flashing? There are two flashing apis: [`flash`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.flash) and [`flash_file`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.flash_file). I'm going to assume you're using the latter. If so, then you could use the former and modify the last couple bytes of your image. E.g.

```
with open('image.bin', 'rb') as f:
    data = f.read()

modified_bytes = [...]
data = data[:len(modified_bytes)] + modified_bytes
```

Then you could call `.flash()` with the target address (same as with `flash_file`), but the modified data above.

hkpeprah, 2022-12-08:
Ability should be available in `v1.0.0` through `flash()`.