square/pylink #137: FileNotFoundError: [Errno 2] No such file or directory: '/opt/SEGGER/JLink_V768b/jlinkarm.so'

skarg, 2022-07-26:
Seems that JLink software for Linux v7.68b (didn't test any other versions) doesn't include a sym link to jlinkarm.so
```
  File "/home/skarg/.local/lib/python3.8/site-packages/pylink/library.py", line 278, in __init__
    self.load(dllpath)
  File "/home/skarg/.local/lib/python3.8/site-packages/pylink/library.py", line 384, in load
    with open(self._path, 'rb') as inputfile:
FileNotFoundError: [Errno 2] No such file or directory: '/opt/SEGGER/JLink_V768b/jlinkarm.so'
skarg@Zephyr-VirtualBox:~/platform$ which JLinkExe
/usr/bin/JLinkExe
skarg@Zephyr-VirtualBox:~/platform$ ls -al /opt/SEGGER/JLink_V768b/
total 64004
drwxr-xr-x  9 root root     4096 Jul 25 18:31 .
drwxr-xr-x  3 root root     4096 Jul 25 18:31 ..
drwxr-xr-x 16 root root     4096 Jul 25 18:31 Devices
drwxr-xr-x  4 root root     4096 Jul 25 18:31 Doc
drwxr-xr-x  3 root root     4096 Jul 25 18:31 ETC
drwxr-xr-x  2 root root     4096 Jul 25 18:31 Firmwares
drwxr-xr-x  2 root root     4096 Jul 25 18:31 GDBServer
-rwxr-xr-x  1 root root  2053768 Jul 22 10:22 JFlashExe
-rwxr-xr-x  1 root root   515280 Jul 22 10:23 JFlashLiteExe
lrwxrwxrwx  1 root root       14 Jul 22 10:25 JFlashSPI_CL -> JFlashSPICLExe
-rwxr-xr-x  1 root root   548008 Jul 22 10:25 JFlashSPICLExe
-rwxr-xr-x  1 root root  1679632 Jul 22 10:23 JFlashSPIExe
-rwxr-xr-x  1 root root   784000 Jul 22 10:25 JLinkConfigExe
-rw-rw-r--  1 root root   136747 Jul 14 12:26 JLinkDevices.xml
-rwxr-xr-x  1 root root   404640 Jul 22 10:25 JLinkExe
lrwxrwxrwx  1 root root       19 Jul 22 10:25 JLinkGDBServer -> JLinkGDBServerCLExe
-rwxr-xr-x  1 root root   556384 Jul 22 10:25 JLinkGDBServerCLExe
-rwxr-xr-x  1 root root   851520 Jul 22 10:25 JLinkGDBServerExe
-rwxr-xr-x  1 root root   490192 Jul 22 10:25 JLinkGUIServerExe
lrwxrwxrwx  1 root root       22 Jul 22 10:25 JLinkLicenseManager -> JLinkLicenseManagerExe
-rwxr-xr-x  1 root root   542256 Jul 22 10:25 JLinkLicenseManagerExe
lrwxrwxrwx  1 root root       20 Jul 22 10:25 JLinkRegistration -> JLinkRegistrationExe
-rwxr-xr-x  1 root root   730480 Jul 22 10:25 JLinkRegistrationExe
lrwxrwxrwx  1 root root       22 Jul 22 10:25 JLinkRemoteServer -> JLinkRemoteServerCLExe
-rwxr-xr-x  1 root root   830616 Jul 22 10:25 JLinkRemoteServerCLExe
-rwxr-xr-x  1 root root  1079024 Jul 22 10:25 JLinkRemoteServerExe
lrwxrwxrwx  1 root root       17 Jul 22 10:25 JLinkRTTClient -> JLinkRTTClientExe
-rwxr-xr-x  1 root root   207912 Jul 22 10:25 JLinkRTTClientExe
lrwxrwxrwx  1 root root       17 Jul 22 10:25 JLinkRTTLogger -> JLinkRTTLoggerExe
-rwxr-xr-x  1 root root   212032 Jul 22 10:25 JLinkRTTLoggerExe
-rwxr-xr-x  1 root root   617928 Jul 22 10:25 JLinkRTTViewerExe
lrwxrwxrwx  1 root root       13 Jul 22 10:25 JLinkSTM32 -> JLinkSTM32Exe
-rwxr-xr-x  1 root root   216064 Jul 22 10:25 JLinkSTM32Exe
lrwxrwxrwx  1 root root       19 Jul 22 10:25 JLinkSWOViewer -> JLinkSWOViewerCLExe
-rwxr-xr-x  1 root root   224312 Jul 22 10:25 JLinkSWOViewerCLExe
-rwxr-xr-x  1 root root   523096 Jul 22 10:25 JLinkSWOViewerExe
-rwxr-xr-x  1 root root   737184 Jul 22 10:24 JMemExe
-rwxr-xr-x  1 root root   285904 Jul 22 10:25 JRunExe
-rwxr-xr-x  1 root root   783720 Jul 22 10:24 JScopeExe
-rwxr-xr-x  1 root root   228368 Jul 22 10:25 JTAGLoadExe
lrwxrwxrwx  1 root root       16 Jul 22 10:25 libjlinkarm.so -> libjlinkarm.so.7
lrwxrwxrwx  1 root root       21 Jul 22 10:25 libjlinkarm.so.7 -> libjlinkarm.so.7.68.2
-rwxr-xr-x  1 root root 18643520 Jul 22 10:25 libjlinkarm.so.7.68.2
lrwxrwxrwx  1 root root       20 Jul 22 10:25 libjlinkarm_x86.so -> libjlinkarm_x86.so.7
lrwxrwxrwx  1 root root       25 Jul 22 10:25 libjlinkarm_x86.so.7 -> libjlinkarm_x86.so.7.68.2
-rwxr-xr-x  1 root root 16961972 Jul 22 10:25 libjlinkarm_x86.so.7.68.2
lrwxrwxrwx  1 root root       14 Jul 22 10:25 libQtCore.so -> libQtCore.so.4
lrwxrwxrwx  1 root root       16 Jul 22 10:25 libQtCore.so.4 -> libQtCore.so.4.8
lrwxrwxrwx  1 root root       18 Jul 22 10:25 libQtCore.so.4.8 -> libQtCore.so.4.8.7
-rwxr-xr-x  1 root root  3215256 Jul 22 10:25 libQtCore.so.4.8.7
lrwxrwxrwx  1 root root       13 Jul 22 10:25 libQtGui.so -> libQtGui.so.4
lrwxrwxrwx  1 root root       15 Jul 22 10:25 libQtGui.so.4 -> libQtGui.so.4.8
lrwxrwxrwx  1 root root       17 Jul 22 10:25 libQtGui.so.4.8 -> libQtGui.so.4.8.7
-rwxr-xr-x  1 root root 11401920 Jul 22 10:25 libQtGui.so.4.8.7
drwxr-xr-x  5 root root     4096 Jul 25 18:31 Samples
drwxr-xr-x  2 root root     4096 Jul 25 18:31 x86
```
I fixed by running:
```
$ sudo ln -s libjlinkarm.so.7.68.2 /opt/SEGGER/JLink_V768b/jlinkarm.so
```

hkpeprah, 2022-08-02:
If you try with the latest version of the PyLink, do you still see this issue?

dottspina, 2022-08-03:
@skarg, does this error happen while using the Zephyr-SDK, and executing a west command, something like `west {flash,debug} --runner jlink` ?
If so, this may be a duplicate of #139, hopefully fixed by #138.

skarg, 2022-08-03:
Yes, this error happens while using Zephyr SDK and executing a west command like "west flash"

dottspina, 2022-08-22:
If this issue is actually a duplicate of #139, should be fixed by pylink 0.14.2.

@skarg, could you confirm the patch also works for you, so that we can close this ?

Thanks.

--
chris

skarg, 2022-08-22:
I don't think I can confirm on the Ubuntu 20.04 system I'm using for ZephyOS builds (unless I'm doing the update wrong):
```
$ sudo rm /opt/SEGGER/JLink_V768b/jlinkarm.so
$ pip install --upgrade pylink
...
Successfully installed pylink-0.3.3
$ pip install -Iv PyLink==0.14.2
ERROR: Could not find a version that satisfies the requirement PyLink==0.14.2 (from versions: 0.1, 0.2, 0.3, 0.3.1, 0.3.2, 0.3.3)
ERROR: No matching distribution found for PyLink==0.14.2
$ pip show PyLink
Name: PyLink
Version: 0.3.3
...
$ west flash
...
-- west flash: using runner jlink
...
FileNotFoundError: [Errno 2] No such file or directory: '/opt/SEGGER/JLink_V768b/jlinkarm.so'
```

dottspina, 2022-08-23:
Thanks for testing.

The software package distributed via PyPI is actually named `pylink-square`, `pylink` itself being a "universal communication interface" that does not seem related to JLink devices in any way. This is a bit confusing when you don't know, I've already done this mistake myself:

- [pylink](https://pypi.org/project/PyLink/) (PyPI)
- [pylink-square](https://pypi.org/project/pylink-square/) (PyPI)

Could you try upgrading with `pip install -U pylink-square` ?

skarg, 2022-08-23:
Success!
```
$ pip install -U pylink-square
...
Successfully installed pylink-square-0.14.2
$ west flash
...
-- west flash: using runner jlink
-- runners.jlink: Flashing file: build/zephyr/.../zephyr.hex
```

hkpeprah, 2022-08-23:
Awesome. Could I mark this resolved for you @skarg ?

skarg, 2022-08-24:
Updating pylink-square to 0.14.2 resolved my issue.

closed 2022-08-24 by skarg