square/pylink #139: Changes to JLINK_SDK_NAME break the jlink runner in Zephyr-RTOS

dottspina, 2022-08-03:
On Linux and macOS, the single constant `JLINK_SDK_NAME` has been used internally by the pylink `Library` class as both:
- the library name parameter for the ctypes API `find_library()`
- the library file name prefix when searching the filesystem for the JLink DLL in the functions `find_library_{linux,darwin}()`

To fix the call to `find_library(JLINK_SDK_NAME)`, the PR #132 has:
- set `JLINK_SDK_NAME` to `'jlinkarm'`
- moved the file name prefix semantic to a new constant `JLINK_SDK_STARTS_WITH`, with value `'libjlinkarm'`

Unfortunately it appears that downstream software has been implemented that rely on the value of `JLINK_SDK_NAME`.

As reported by @asarbs ([here](https://github.com/square/pylink/pull/132#discussion_r935327549)), this is the case of the `jlink` runner in Zephyr-SDK.

I can reproduce a similar error, where this runner will attempt to load a library file named `jlinkarm.so` instead of `libjlinkarm.so`:

```
$ west flash --runner jlink 
-- west flash: using runner jlink
Traceback (most recent call last):
[...]
  File "/mnt/platform/zephyr-rtos/workspaces/zephyr-upstream/.venv/lib64/python3.10/site-packages/pylink/library.py", line 384, in load
    with open(self._path, 'rb') as inputfile:
FileNotFoundError: [Errno 2] No such file or directory: '/mnt/platform/segger/JLink_Linux_V766e_x86_64/jlinkarm.so'
```

I'm quite sure it's also the root cause of #137.

I initially thought that we might fix this downstream, considering the `west` command implementation should not rely on pylink *internal* constant `Library.JLINK_SDK_NAME`.

But we may also consider `JLINK_SDK_NAME` is somewhat a public symbol (for e.g. it does not start with `_`, which AFAICT would have been the strongest Python hint to tell it's private), and that we, **I**, simply should not break any downstream software that happens to rely on in its value.
At the end of the day this may be the only acceptable approach, a bit like *we don't break userspace*.

The PR #138 restores `JLINK_SDK_NAME` former value and semantic, without reverting PR #132. 

Thanks.

--
chris

asarbs, 2022-08-03:
Hi i have fixed this issue on my side by, so mayby this can be fixed in zephyr-os/west 

```
diff --git a/scripts/west_commands/runners/jlink.py b/scripts/west_commands/runners/jlink.py
index 579bb81212..d6aad8065a 100644
--- a/scripts/west_commands/runners/jlink.py
+++ b/scripts/west_commands/runners/jlink.py
@@ -152,12 +152,12 @@ class JLinkBinaryRunner(ZephyrBinaryRunner):
             if plat.startswith('win32'):
                 libname = Library.get_appropriate_windows_sdk_name() + '.dll'
             elif plat.startswith('linux'):
-                libname = Library.JLINK_SDK_NAME + '.so'
+                libname = Library.JLINK_SDK_STARTS_WITH + '.so'
             elif plat.startswith('darwin'):
-                libname = Library.JLINK_SDK_NAME + '.dylib'
+                libname = Library.JLINK_SDK_STARTS_WITH + '.dylib'
             else:
                 self.logger.warning(f'unknown platform {plat}; assuming UNIX')
-                libname = Library.JLINK_SDK_NAME + '.so'
+                libname = Library.JLINK_SDK_STARTS_WITH + '.so'
 
             lib = Library(dllpath=os.fspath(Path(self.commander).parent /
                                             libname))
```

dottspina, 2022-08-03:
> Hi i have fixed this issue on my side by, so mayby this can be fixed in zephyr-os/west
>
> ```
> diff --git a/scripts/west_commands/runners/jlink.py b/scripts/west_commands/runners/jlink.py
> index [...]

@asarbs, this is asking the Zephyr peoples for swapping symbols, arguing that the new name better describes the semantic. Which is true, but I'm definitely inclined to think it's a weak reason to break downstream, and that we should rather swap the symbols ourselves. PR #138 does exactly this.

asarbs, 2022-08-03:
Ok, i needed some quick fix, to do my work this was the easiest for me, like 3 minutes of work.

dottspina, 2022-08-03:
> Ok, i needed some quick fix, to do my work this was the easiest for me, like 3 minutes of work.

Fine. Thanks for reporting and narrowing the issue.

dottspina, 2022-08-22:
Should be fixed by pylink 0.14.2 (Works for me TM).

@asarbs, could you confirm it's also fixed for you, so that we can close this issue ?

Thanks.

--
chris

dottspina, 2023-01-24:
I'll close this:
- OP (@asarbs) did not confirm he could reproduce the issue with pylink 0.14.2         
- I'm a frequent user of JLink/Zephyr myself, subscribed to most of Zephyr mailing lists, and follow their issue tracker: no complain regarding the jlink west runner
 
Feel free to reopen.

Thanks.

--
chris

closed 2023-01-24 by dottspina