square/pylink #144: target_connected() returns false after executing connect() without issues

raulgotor, 2022-09-02:
Hello,

**Environment**

macOS Catalina 10.15.7, Python 3.10, JLink 7.20 and PyLink 0.14.2.

I'm trying to use PyLink with JLink and NXP QN9080 MCU. Apparently I can connect (my dev board has a LED that turns on when connected to JLink), but execution of any function that internally calls `target_connected()` fails, as this method is returning false.

**Expected behaviour**

After opening a connection with JLink and connecting to the target, methods communicating with the target work properly. Interrogating for `target_connected` returns true.

**Actual behaviour**

After opening a connection with JLink and connecting to the target, methods communicating with the target report a `JLinkException: target is not connected` error. 

**Steps to reproduce**

```python
In [1]: import pylink
In [2]: jlink = pylink.JLink()
In [3]: jlink.open()
In [4]: jlink.connect("QN9080C")
In [5]: jlink.target_connected()
Out[5]: False
```

For instance, calling `jlink.flash(...,...)` now produces:

```python
~/....../env/lib/python3.8/site-packages/pylink/jlink.py in wrapper(self, *args, **kwargs)
    170             """
    171             if not self.target_connected():
--> 172                 raise errors.JLinkException('Target is not connected.')
    173             return func(self, *args, **kwargs)
    174         return wrapper

JLinkException: Target is not connected.
```

**Aditional information**
I tried to call directly the JLink library API used at `.connect()` and at `.target_connected()`. It produces the following output:

```python
In [6]: jlink._dll.JLINKARM_IsConnected()
Out[6]: 0

In [7]: jlink._dll.JLINKARM_EMU_IsConnected()
Out[7]: 1
```

Tried with a JLink Pro and with a JLink Trace with no luck. Didn't tried on a different computer.

Any clues? Thanks!

hkpeprah, 2022-09-09:
Sorry about the late response here. Hm. I'm not sure. If you run with verbose logging, does it show you anything?
```
import logging

logging.basicConfig(level=logging.DEBUG)
```

raulgotor, 2022-09-12:
Hey, thanks for your message: below the logging output to `open()` and `connect()`:

```
In [8]: jlink.open()
DEBUG:pylink.jlink:TCA9DDC0 059:341.862
DEBUG:pylink.jlink:Firmware: J-Trace PRO V3 Cortex-M compiled Jul 15 2022 11:23:45
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:342.341
DEBUG:pylink.jlink:Firmware: J-Trace PRO V3 Cortex-M compiled Jul 15 2022 11:23:45
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:348.122
DEBUG:pylink.jlink:Hardware: V3.00
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:348.334
DEBUG:pylink.jlink:S/N: 753002218
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:348.450
DEBUG:pylink.jlink:OEM: SEGGER
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:348.557
DEBUG:pylink.jlink:Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:351.503
DEBUG:pylink.jlink:TELNET listener socket opened on port 19021
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:351.742
DEBUG:pylink.jlink:WEBSRV Starting webserver
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:351.887
DEBUG:pylink.jlink:WEBSRV Webserver running on local port 19080
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:351.989
DEBUG:pylink.jlink:- 16.196ms returns "O.K."

DEBUG:pylink.jlink:TCA9DDC0 059:352.129
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog

DEBUG:pylink.jlink:TCA9DDC0 059:352.221
DEBUG:pylink.jlink:JLINK_SetHookUnsecureDialog(...)
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 059:352.309
DEBUG:pylink.jlink:- 0.210ms returns 0
```

```
In [10]: jlink.connect("QN9080C")
DEBUG:pylink.jlink:TCA9DDC0 084:910.025
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 084:910.168
DEBUG:pylink.jlink:- 0.196ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 084:910.307
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 084:910.393
DEBUG:pylink.jlink:- 0.122ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 084:910.490
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 084:910.603
DEBUG:pylink.jlink:- 0.151ms returns TRUE

DEBUG:pylink.jlink:TCA9DDC0 084:910.778
DEBUG:pylink.jlink:JLINK_DEVICE_GetIndex(sDeviceName = QN9080C)

DEBUG:pylink.jlink:TCA9DDC0 084:910.975
DEBUG:pylink.jlink:- 0.239ms returns 4538

DEBUG:pylink.jlink:TCA9DDC0 084:911.099
DEBUG:pylink.jlink:JLINK_DEVICE_GetInfo(DeviceIndex = -1)

DEBUG:pylink.jlink:TCA9DDC0 084:911.173
DEBUG:pylink.jlink:- 0.107ms returns 8278

DEBUG:pylink.jlink:TCA9DDC0 084:911.323
DEBUG:pylink.jlink:JLINK_DEVICE_GetInfo(DeviceIndex = 4538)

DEBUG:pylink.jlink:TCA9DDC0 084:911.454
DEBUG:pylink.jlink:- 0.173ms returns 0

DEBUG:pylink.jlink:TCA9DDC0 084:911.630
DEBUG:pylink.jlink:JLINK_ExecCommand("Device = QN9080C", ...).

INFO:pylink.jlink:Device "QN9080C" selected.
DEBUG:pylink.jlink:TCA9DDC0 084:912.392
DEBUG:pylink.jlink:Device "QN9080C" selected.
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 084:912.789
DEBUG:pylink.jlink:- 1.129ms returns 0x00

DEBUG:pylink.jlink:TCA9DDC0 084:912.892
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 084:912.959
DEBUG:pylink.jlink:- 0.097ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 084:913.024
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 084:913.084
DEBUG:pylink.jlink:- 0.090ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 084:913.147
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 084:913.206
DEBUG:pylink.jlink:- 0.089ms returns TRUE

DEBUG:pylink.jlink:TCA9DDC0 084:913.301
DEBUG:pylink.jlink:JLINK_SetSpeed(0)

INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 084:915.496
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:

INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 085:069.733
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 085:070.501
DEBUG:pylink.jlink:- 157.374ms

DEBUG:pylink.jlink:TCA9DDC0 085:070.806
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 085:070.979
DEBUG:pylink.jlink:- 0.248ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 085:071.141
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 085:071.295
DEBUG:pylink.jlink:- 0.224ms returns TRUE

DEBUG:pylink.jlink:TCA9DDC0 085:073.920
DEBUG:pylink.jlink:JLINK_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 085:074.115
DEBUG:pylink.jlink:- 0.242ms returns FALSE

DEBUG:pylink.jlink:TCA9DDC0 085:074.243
DEBUG:pylink.jlink:JLINK_Connect()

INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 085:077.860
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:

INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 085:235.320
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:

DEBUG:pylink.jlink:TCA9DDC0 085:236.104
DEBUG:pylink.jlink:- 162.035ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 085:236.410
DEBUG:pylink.jlink:JLINK_IsOpen()

DEBUG:pylink.jlink:TCA9DDC0 085:236.577
DEBUG:pylink.jlink:- 0.250ms returns 0x01

DEBUG:pylink.jlink:TCA9DDC0 085:236.715
DEBUG:pylink.jlink:JLINK_EMU_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 085:236.811
DEBUG:pylink.jlink:- 0.144ms returns TRUE

DEBUG:pylink.jlink:TCA9DDC0 085:236.912
DEBUG:pylink.jlink:JLINK_IsConnected()

DEBUG:pylink.jlink:TCA9DDC0 085:237.008
DEBUG:pylink.jlink:- 0.142ms returns FALSE
```

hkpeprah, 2022-10-06:
It sort of seems like the `Connect` doesn't succeed:
```
INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 085:077.860
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:

INFO:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:TCA9DDC0 085:235.320
DEBUG:pylink.jlink:TotalIRLen = ?, IRPrint = 0x..000000000000000000000000
DEBUG:pylink.jlink:
```

Sschumac, 2023-10-26:
I know this is old.. but, im also experiencing this issue.