square/pylink #154: Programming without erasing

denravonska, 2022-11-23:
We have a system where the bootloader and some additional data need to be retained on the flash while the application can be erased. Using chip erase will nuke all of it.

I've been digging through the SDK documentation and it looks like it's possible to do using Begin- and EndDownload. It would be something like this:

```Python
data = open(path, mode="rb").read()
self._dll.JLINKARM_BeginDownload(0)
self._dll.JLINKARM_WriteMem(addr, len(data), data)
self._dll.JLINKARM.EndDownload()
```

I could give this a shot, but how would you like it implemented? Integrated into the current flash operations (`flash` and `flash_file`) or as a new function? I don't think it's possible to get a progress bar in that case, so a new function might be in order.

hkpeprah, 2022-11-23:
I think modifying `flash()` would be the best option, and could be a major change (bump of the major version).

denravonska, 2022-11-25:
After doing some tests it seems that progress does actually work with `WriteMem`. 

```
b'Erase' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0%  
b'Erase' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0%  
b'Compare' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0% 
b'Compare' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0% 
b'Erase' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0% 
b'Program' |████████████████████████████████████████████████████████████████████████████████████████████████████| 100.0% 
```

PR #155 (2022-12-08): Switch to read-modify-write flashing.
This enables programming the chip without having to erase the entire contents. Instead, JLink will erase where needed while retaining the data that's within the same sectors but outside of the specified write area.

I was only able to run some of the Behave tests as I don't have access to the hardware.

Closes #154, #148
closed 2022-12-08 by hkpeprah