square/pylink #157: Flashing .hex files

JonathanCBU, 2023-01-05:
Hello, 

I need to flash .hex files programmatically, and I am already using pylink for RTT viewer stuff so I'd like to keep using the same library. 

Now, JFlashLite can flash a hex file using the GUI without any issue, but the [flash_file](https://github.com/square/pylink/blob/master/pylink/jlink.py#L2212) method doesn't seem to work (I'm assuming that's because of the address encoding within hex files). 

When attempting to flash using `Jlink.flash_file(<hex_file>, 0)` I see the UI appear, but it never moves past erasing, and it always crashes with `JLinkFlashException: Programming error occured.`

Is there a recommended method for flashing hex files? If not, Is the next best thing parsing the file and flashing it in byte chunks?

JonathanCBU, 2023-01-05:
Ahhh I see, and then I should be able to flash that whole set of data starting at adder 0?

hkpeprah, 2023-01-05:
Yup!

JonathanCBU, 2023-01-06:
That method definitely made a flash-able bin array (although I needed one more step of `data_list = data.tolist()`). 

Unfortunately though, the `.erase()` function does not seem to reliably work reliably. It did work a few times, but I cannot pinpoint what is changing between attempts. Per [this discussion](https://github.com/square/pylink/issues/135) I am making sure the Jlink is halted before erasing, but I do not always see that erasing happens. 

Here is how I am setting up my Jlink.

```
import pylink
import intelhex

jlink = pylink.JLink()
jlink.open(<jlink_sn>)
jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
jlink.connect(<chip_name>)

ih = intelhex.IntelHex()
ih.fromfile(<hex_file>, format='hex')
data = ih.tobinarray()
d_list = data.tolist()

# in an python session I would confirm jlink.halted() returns true before continuing
jlink.halt()

current_mem = jlink.memory_read(0, len(d_list))

# here I would see the JLink UI flash on screen for less than a second and the function would return 0
jlink.erase() 

 # should be a list of 255 but instead returns the same thing as current_mem
erased_mem = jlink.memory_read(0, len(d_list))

# fails if erased_mem is not a list of 255 (fails with "ValueError: Invalid error code: -3")
jlink.flash(d_list, 0)
```

hkpeprah, 2023-01-09:
Do you have the debugging logs? The logs that print to the Python logger? Sometimes, the CPU isn't halted properly due to something else on the device waking it up.

closed 2023-10-12 by JonathanCBU