square/pylink #158: Cortex-M33 not connecting as expected

s-bruce13, 2023-01-06:
Hello Im having an issue with CORTEX-M33 devices. I have tested using other devices such as CORTEX-M4 and those connect as expected. Im on Windows 10. Python version 3.11.1 Pylink 1.0.0 (I've tested using Python2.7 as well with same results)

>>> import pylink
>>> jlink=pylink.JLink()
>>> jlink.open(600110070)
>>> jlink.product_name
'SEGGER J-Link ARM'
>>> jlink.oem
>>> jlink.opened()
True
>>> jlink.connected()
True
>>> jlink.target_connected()
False
>>> jlink.connect('CORTEX-M33')
>>> jlink.target_connected()
False

I can connect to target device just fine if I use Jlink directly:
Connecting to target via SWD
Found SW-DP with ID 0x6BA02477
DPIDR: 0x6BA02477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[3]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x84770001)
AP[1]: AHB-AP (IDR: 0x84770001)
AP[2]: JTAG-AP (IDR: 0x002A0000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FE000
CPUID register: 0x410FD213. Implementer code: 0x41 (ARM)
Feature set: Mainline
Found Cortex-M33 r0p3, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
Security extension: implemented
Secure debug: enabled
CoreSight components:
ROMTbl[0] @ E00FE000
[0][0]: E00FF000 CID B105100D PID 000BB4C9 ROM Table
ROMTbl[1] @ E00FF000
[1][0]: E000E000 CID B105900D PID 000BBD21 DEVARCH 47702A04 DEVTYPE 00 Cortex-M33
[1][1]: E0001000 CID B105900D PID 000BBD21 DEVARCH 47701A02 DEVTYPE 00 DWT
[1][2]: E0002000 CID B105900D PID 000BBD21 DEVARCH 47701A03 DEVTYPE 00 FPB
[1][3]: E0000000 CID B105900D PID 000BBD21 DEVARCH 47701A01 DEVTYPE 43 ITM
[0][1]: E0040000 CID B105900D PID 000BBD21 DEVARCH 00000000 DEVTYPE 11 TPIU
Memory zones:
  Zone: Default Description: Default access mode
Cortex-M33 identified.

Is there anything to change in order to test this connection. Ive tried modifying speed as well but behavior is the same and I have tried several devices that are CORTEX-M33 based.

hkpeprah, 2023-01-09:
Have you tried with the exact device as opposed to passing in `Cortex-M33`? E.g., the `STM32L552ZC` is a `Cortex-M33`, so passing `STM32L552CZ` instead of `Cortex-M33` when you call `.connect()`?

s-bruce13, 2023-01-09:
Hello the end device I am working with is not currently in supported jlink list, I can connect to it using CORTEX-M33.. However I have tested with other devices such as LPC55S69 and it does not connect either as the specific device name or CORTEX-M33.

Additionally I tested with K64(specific device name) and CORTEX-M4 and both of these options connect successfully.

s-bruce13, 2023-01-09:
ok that is a good tip i can take a look into. I did try to run some coresight functions that are done in the regular jlink connect script, but seems that the coresight write/read calls the .connect function as i will get an error like "Target not connected"

I've tried to connect through jlink and jump back to the pylink functions once I knew jlink had connected, but this did not work. 

Do you have any suggestion as to how i may enable the connection prior to being able to succesfully connect to target?

s-bruce13, 2023-01-09:
Thanks for information. I attempted to configure the coresight. However it causes the jlink to disconnect. Do you have any examples using a CORTEX-M33 device that i can use for reference? Here is my output:

>>> import pylink
>>> jlink = pylink.JLink()
>>> jlink.open(600110070)
>>> jlink.product_name
'SEGGER J-Link ARM'
>>> jlink.oem
>>> jlink.opened()
True
>>> jlink.connected()
True
>>> jlink.coresight_configure(0,0,0,0,4)
Communication timed out: Requested 4 bytes, received 0 bytes !
>>> jlink.connected()
False
>>> jlink.opened()
True
>>> jlink.close()
>>> jlink.opened()
False
>>> jlink.connected()
False
>>> jlink.open(600110070)
>>> jlink.product_name
'SEGGER J-Link ARM'
>>> jlink.oem
>>> jlink.opened()
True
>>> jlink.connected()
True
>>> jlink.coresight_configure(0,0,0,0,4)
>>> jlink.connected()
False
>>> jlink.opened()
True

s-bruce13, 2023-01-09:
I was able to connect using a line i found from another issue:

I added 
jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
prior to calling .connect()

Thanks for your help!

closed 2023-01-09 by s-bruce13

enjeru15, 2023-10-20:
Hi @s-bruce13 , when you implemented the jlink.set_tif, did it work already with Cortex-M33 without the coresight_configure() and coresight_write() functions? 'Cause until now, I'm having the same issue connecting with Cortex-M33. I think I'm not properly setting the core ー application core or net core.

enjeru15, 2023-10-20:
Are you using a dual core? If so, how are you setting it using the jlink functions?

s-bruce13, 2023-10-20:
> Are you using a dual core? If so, how are you setting it using the jlink functions?

im not using dual core just single.