square/pylink #159: "Unsupported device selected" error with loader strings

multiplemonomials, 2023-01-11:
I've been testing a chip that requires a loader string as part of the device argument.  For example, on the command line, I'd pass `-Device MIMXRT1062xxx5B?BankAddr=0x60000000&Loader=QSPI`.  However, when I try the corresponding code in pylink:

```python
jlink.connect("MIMXRT1062xxx5B?BankAddr=0x60000000&Loader=QSPI")
```
it dies with an "Unsupported device selected" exception from JLink.get_device_index().  It looks like this is because get_device_index() expects an un-adorned string, e.g. "MIMXRT1062xxx5B".

As a workaround, I figured out that if I change the `connect()` implementation to pass only the part of chip_name before the '?' character to get_device_index(), everything works.  The loader string correctly gets passed via code after that, and I can flash and run code.

multiplemonomials, 2023-01-11:
Hmm, on further testing, turns out that flashing doesn't actually work with the above workaround.  I have to run
```python
jlink.exec_command('DEVICE_SelectLoader BankAddr=0x60000000 Loader=QSPI')
```
before calling connect().  Maybe connect() could be updated to parse out the loader string and pass it via DEVICE_SelectLoader?

Also, discovered that on this device, you have to call `jlink.reset(halt=False)` immediately before calling flash_file.  Not sure what's up with that...

hkpeprah, 2023-01-11:
I imagine we could parse what is after the `?`, though I think we would have to make sure that no device name contains a query string. Something like:
```
device_list = name.split('?')
device_name = next(device_list)
device_loader = next(device_list, None)

<...>

if device_loader is not None:
    self.exec_command(f'" ".join({device_loader}.split("&"))')
```

Thoughts?

multiplemonomials, 2023-01-11:
Hmm, I think it might be possible to have a loader string with multiple banks and loaders for them?  I'm not sure what the syntax would look like though.  Maybe you'd end up with something like:

```
MIMXRT1062xxx5B?BankAddr=0x60000000&Loader=QSPI&BankAddr=0x70000000&Loader=OSPI
```

The parser would have to handle this too.  And we'd need to find an example of a real loader string like this to be sure.

bbilas, 2023-02-09:
I encountered the same problem during the development using the board with `MIMXRT1062` MCU. The command that I use to flash is `JLinkExe -nogui 1 -if swd -speed auto -device 'MIMXRT1062xxx6A?BankAddr=0x60000000&Loader=QSPI' -CommanderScript /tmp/runner.jlink -nogui 1` where `runner.jlink` is 

<pre>
ExitOnError 1
r
loadfile "/tmp/build/zephyr/zephyr.hex"
g
writeDP 1 0
readDP 1
q
</pre>

Is there any PR that adds support for this?

hkpeprah, 2023-02-09:
There is no PR yet that adds this support, unfortunately.