square/pylink #160: Invalid error code: -11 from rtt_read()

multiplemonomials, 2023-01-19:
Hi, I've been using pylink on a couple boards and it's been working pretty well so far!  However, today one of my boards got in a bad state where RTT can't be used with it.  My code has:
```py
jlink.rtt_read(0, 1024)
```

and the stacktrace I get is:
```
  File "/home/jsmith/.local/lib/python3.10/site-packages/pylink/jlink.py", line 142, in wrapper
    return func(self, *args, **kwargs)
  File "/home/jsmith/.local/lib/python3.10/site-packages/pylink/jlink.py", line 5111, in rtt_read
    raise errors.JLinkRTTException(bytes_read)
  File "/home/jsmith/.local/lib/python3.10/site-packages/pylink/errors.py", line 38, in __init__
    message = self.to_string(code)
  File "/home/jsmith/.local/lib/python3.10/site-packages/pylink/enums.py", line 264, in to_string
    return super(JLinkRTTErrors, cls).to_string(error_code)
  File "/home/jsmith/.local/lib/python3.10/site-packages/pylink/enums.py", line 96, in to_string
    raise ValueError('Invalid error code: %d' % error_code)
ValueError: Invalid error code: -11
``` 

It looks like the J-Link DLL is returning error code -11 but this library doesn't know how to interpret that?  btw, it's worked fine with this same board and script in the past.

hkpeprah, 2023-02-22:
Sorry, I do not know what that error code is. The logs are usually a better descriptor, so if you don't have logging enabled, then I would enable it:
```
import logging

logging.basicConfig(level=logging.DEBUG)
```

vincentdman, 2023-08-17:
Hey i have the same problem. 
It seems to be a usage error. i can reproduce it by attaching a jlink to my mcu. 
have the rtt_read() running in a thread.
then connect with the wrong device selected using jlink.connect.
then connect with the right device and program that device (i used a script for programming called by subprocess). 
It gives the same exception traceback.
So i fixed it by first disconnecting before connecting to a new device. 
I hope this helps someone

diggit, 2024-10-22:
I am occasionally seeing this error code `-11` when I have pylink with rtt running and at the same time I detach JLink gdb server from the target.

hkpeprah, 2024-10-22:
I think you can use RTT and GDB at the same time. It's possible that detaching from GDB server is sending some command that interrupts the debugger when you're reading from RTT. If you can run with logging enabled, it should provide more information.