square/pylink #166: The temporary copy of library is not deleted when Library object is released

chanqueo, 2023-02-14:
I use pylink-square Python module with pytest to run tests on almost 100 devices at a time and several time a day.
In some conditions I couldn't list easily here, the temporary copy of J-Link library is not deleted at the end of the tests.
(probably because is is done with Library __del__ method?)
Each copy represents about 15MB on disk, this will then represent more than 1 GB after several days or weeks.

hkpeprah, 2023-02-22:
Sorry about the late response here. The file should be cleaned up as a part of the `__del__` (the destructor) when the class instance is garbage collected. Is it possible that your tests are exiting without allowing the Python garbage collector to run? This workaround was put in place because an early version of the library did not allow for multiple device communication from the same machine at the same time. That has since been patched, so we can probably put a version guard around the copy, but would have to be careful in how we do that; e.g. copy the library first, check the version, if it is high enough, then delete the copy and use the original.

hkpeprah, 2023-04-19:
Will take a look!

hkpeprah, 2023-05-09:
`v1.1.0` should have this change in it now; sorry for the delay, and thank you for the patch!

closed 2025-02-07 by hkpeprah