square/pylink #167: Unable to flash 2 different MCUs in a row

bbilas, 2023-02-21:
I have a case where I want to flash 2 different MCUs using the same JLink debugger connected via USB and 1 instance of PyLink lib. I have 1 MIMXRT1062XXX6A and 2 CC1352 MCUs connected via analog muxes to the JLink where I'm able to flash only the first one, the second one stucks on `flash` method. When I change the order of MCUs I can successfully flash the first one but the second one which was working fine is nonusable.

That's the log from the "second" MCU which is stucking on flashing.

<pre>
Device "MIMXRT1062XXX6A" selected.	
InitTarget() start	
InitTarget() end	
Found SW-DP with ID 0x0BD11477	
DPIDR: 0x0BD11477	
CoreSight SoC-400 or earlier	
Scanning AP map to find all available APs	
AP[1]: Stopped AP scan as end of AP map has been reached	
AP[0]: AHB-AP (IDR: 0x04770041)	
Iterating through AP map to find AHB-AP to use	
AP[0]: Core found	
AP[0]: AHB-AP ROM base: 0xE00FD000	
CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)	
Found Cortex-M7 r1p1, Little endian.	
FPUnit: 8 code (BP) slots and 0 literal slots	
CoreSight components:	
ROMTbl[0] @ E00FD000	
[0][0]: E00FE000 CID B105100D PID 000BB4C8 ROM Table	
ROMTbl[1] @ E00FE000	
[1][0]: E00FF000 CID B105100D PID 000BB4C7 ROM Table	
ROMTbl[2] @ E00FF000	
[2][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7	
[2][1]: E0001000 CID B105E00D PID 000BB002 DWT	
[2][2]: E0002000 CID B105E00D PID 000BB00E FPB-M7	
[2][3]: E0000000 CID B105E00D PID 000BB001 ITM	
[1][1]: E0041000 CID B105900D PID 001BB975 ETM-M7	
[1][2]: E0042000 CID B105900D PID 004BB906 CTI	
[0][1]: E0040000 CID B105900D PID 000BB9A9 TPIU-M7	
[0][2]: E0043000 CID B105F00D PID 001BB101 TSG	
Cache: Separate I- and D-cache.
I-Cache L1: 32 KB, 512 Sets, 32 Bytes/Line, 2-Way	
D-Cache L1: 32 KB, 256 Sets, 32 Bytes/Line, 4-Way
</pre>

I have verified hardware using JFlashLite app where I can flash all of them without any problems.

hkpeprah, 2023-02-22:
Is it possible that you need to select the different MCUs on the JTAG scan chain manually?

bbilas, 2023-02-22:
> Is it possible that you need to select the different MCUs on the JTAG scan chain manually?

Would you mind providing some more details? After every flash I close jlink connection and I do something like

<pre>
    def __connect_jlink(self, device_id, jtag):
        self.__jlink.open()
#        self.__jlink.disable_dialog_boxes()
        if jtag is True:
            self.__jlink.set_tif(pylink.enums.JLinkInterfaces.JTAG)
        else:
            self.__jlink.set_tif(pylink.enums.JLinkInterfaces.SWD)
        self.__jlink.connect(f'{device_id}', verbose=True)
</pre>

and the connection itself looks good. The issue is somewhere in the flash step.

hkpeprah, 2023-03-08:
I assume that the devices are all on the same JTAG scan chain. It's possible that the flashing application is configuring that for you, whereas with this library, you would have to do it yourself. I would take a look at [`jtag_configure`](https://pylink.readthedocs.io/en/latest/pylink.html#pylink.jlink.JLink.jtag_configure).

bbilas, 2023-07-06:
Sorry for the **really** late answer but you were right. In case of connected multiple MCUs, there is a need to call `jtag_configure`.

closed 2023-07-06 by bbilas