square/pylink #170: raise errors.JLinkException('Target is not connected.')

TheJester12121, 2023-03-23:
Hi all,

I'm currently figuring out what is wrong with my code. Im trying Pylink, but seem to be unable to do for instance a reset.
The board im using is a Nordic nRF5340-DK. It is a dual core microcontroller.

```

import pylink
jlink = pylink.JLink()
jlink.open("000601017523")

print("Product name : {}".format(jlink.product_name))
print("Debugger version : {}".format(jlink.version))
print("Debugger serial : {}".format(jlink.serial_number))

if not jlink.oem:
        print("Original Segger device attached")
else:
        print("Segger OEM : {}".jlink.oem)

print(jlink.supported_device(0))
print(jlink.supported_device(1))

print("Jlink self test result : {}".format(jlink.test()))
print("Power on")
try:
        jlink.power_on()
except pylink.JLinkException as e:
        print("pylink.JLinkException {}".format(str(e)))

print("Try to connect")

try:
        jlink.connect("nRF5340_xxAA_NET", speed='auto', verbose=True)
except pylink.JLinkException as e:
        print("pylink.JLinkException {}".format(str(e)))

print("after connect, what is the status")
print("Connected: {}".format(jlink.connected()))
print("Target connected : {}".format(jlink.target_connected()))

print("try reset")

jlink.reset(ms=1000,halt=True)
jlink.erase()

jlink.power_off()

```

When executed, i get the following:

```
Product name : SEGGER J-Link ARM
Debugger version : 7.66g
Debugger serial : 601017523
Original Segger device attached
ARM7 <Core Id. 134217727, Manu. Unspecified>
ARM9 <Core Id. 167772159, Manu. Unspecified>
Jlink self test result : True
Power on
Try to connect
after connect, what is the status
Connected: True
Target connected : False
try reset
Traceback (most recent call last):
  File "/home/martijn/playground/t.py", line 37, in <module>
    jlink.reset(ms=1000,halt=True)
  File "/home/martijn/.local/lib/python3.10/site-packages/pylink/jlink.py", line 172, in wrapper
    raise errors.JLinkException('Target is not connected.')
pylink.errors.JLinkException: Target is not connected.
```

In my view this means that:

- power on didnt throw
- connect didnt throw
- connected shows True

now the question: why is the target not connected ? 

Am i missing some vital step from the API ? 

Btw, the board, together with the external debugger and the JTAG to SWD cable combination does work when used in conjunction with the Nordic tools 'nrfjprog' which in turn just uses the Jlink executables and libs.

Any help is much appreciated.

hkpeprah, 2023-03-23:
I would suggest trying these troubleshooting steps here: https://pylink.readthedocs.io/en/latest/troubleshooting.html#unspecified-error-during-connect

TheJester12121, 2023-03-23:
Thanks!!

Yes, seen that one indeed.
But my error doesn't say: JLinkException: Unspecified error, it says JLinkException: Target is not connected..
The whole page you mentioned is related to Unspecified error; 
I have tried the other things mentioned by the unspecified error related to connect, but that didnt get me anywhere.

hkpeprah, 2023-03-29:
The issue is likely in the setup somewhere. Are you sure the target device name is correct?

hkpeprah, 2023-05-09:
Closing issue as stale.

closed 2023-05-09 by hkpeprah