square/pylink #175: Tests fail on some architectures (using unittest.mock)

johanneskastl, 2023-05-14:
I am trying to package for openSUSE this as it is a dependency of pynitrokey.

As openSUSE tries to remove use of python-mock, I simply switched all tests to unittest.mock:

```bash
find tests -type f -exec sed -i 's/from mock/from unittest.mock/g' {} +                                                                                                                      
find tests -type f -exec sed -i 's/import mock/import unittest.mock as mock/g' {} +
```

Worked like  charm (and might be usable for #149), as the tests pass successfully on x86_64.

However there are some that fail on non-x86:

## armv7l

- tests/unit/test_library.py::TestLibrary::test_initialize_windows

## i586

- tests/unit/test_library.py::TestLibrary::test_initialize_windows

## ppc64

- tests/unit/unlockers/test_unlock_kinetis.py::TestUnlockKinetis::test_unlock_kinetis_read_fail
- tests/unit/unlockers/test_unlock_kinetis.py::TestUnlockKinetis::test_unlock_kinetis_success

## s390x

- tests/unit/unlockers/test_unlock_kinetis.py::TestUnlockKinetis::test_unlock_kinetis_read_fail
- tests/unit/unlockers/test_unlock_kinetis.py::TestUnlockKinetis::test_unlock_kinetis_success

## Errors

```bash
[   32s] =================================== FAILURES ===================================
[   32s] _____________________ TestLibrary.test_initialize_windows ______________________
[   32s] 
[   32s] self = <tests.unit.test_library.TestLibrary testMethod=test_initialize_windows>
[   32s] mock_ctypes = <MagicMock name='ctypes' id='4116548240'>
[   32s] mock_find_library = <MagicMock name='find_library' id='4116458680'>
[   32s] mock_open = <MagicMock name='open' id='4116527232'>
[   32s] 
[   32s]     @mock.patch('sys.platform', new='windows')
[   32s]     @mock.patch('pylink.library.open')
[   32s]     @mock.patch('os.remove', new=mock.Mock())
[   32s]     @mock.patch('tempfile.NamedTemporaryFile', new=mock.Mock())
[   32s]     @mock.patch('ctypes.util.find_library')
[   32s]     @mock.patch('pylink.library.ctypes')
[   32s]     def test_initialize_windows(self, mock_ctypes, mock_find_library, mock_open):
[   32s]         """Tests creating a library on a Windows machine.
[   32s]     
[   32s]         Args:
[   32s]           self (TestLibrary): the ``TestLibrary`` instance
[   32s]           mock_ctypes (Mock): a mocked version of the ctypes library
[   32s]           mock_find_library (Mock): mock for mocking the
[   32s]             ``ctypes.util.find_library()`` call
[   32s]           mock_open (Mock): mock for mocking the call to ``open()``
[   32s]     
[   32s]         Returns:
[   32s]           ``None``
[   32s]         """
[   32s]         mock_windll = mock.Mock()
[   32s]         mock_windll.__getitem__ = mock.Mock()
[   32s]     
[   32s]         mock_cdll = mock.Mock()
[   32s]         mock_cdll.__getitem__ = mock.Mock()
[   32s]     
[   32s]         mock_ctypes.windll = mock_windll
[   32s]         mock_ctypes.cdll = mock_cdll
[   32s]         mock_find_library.return_value = self.lib_path
[... 127 lines cut ...]
[   25s]             ack,   # Erase Command ACK
[   25s]             0x1,   # Erase Command Parity
[   25s]             ack,   # Erase Command ACK
[   25s]             0x0,   # Erase Command Parity
[   25s]             ack,   # Flash Erase Command ACK
[   25s]             0x1,   # Flash Erase Command Parity
[   25s]             ack,   # Flash Erase Command ACK
[   25s]             0x0,   # Flash Erase Command Parity
[   25s]         ]
[   25s]     
[   25s]         # ACK
[   25s]         mock_jlink.swd_write.return_value = 0
[   25s]     
[   25s]         # Status and Parity
[   25s]         mock_jlink.swd_read8.side_effect = mocked_data.status_and_parity_data
[   25s]     
[   25s]         # Data
[   25s]         mock_jlink.swd_read32.side_effect = mocked_data.read_data
[   25s]     
[   25s]         res = unlock.unlock_kinetis(mock_jlink)
[   25s] >       self.assertTrue(res)
[   25s] E       AssertionError: False is not true
[   25s] 
[   25s] tests/unit/unlockers/test_unlock_kinetis.py:221: AssertionError
```

hkpeprah, 2023-05-18:
We can probably switch to `unittest.mock` in dropping support for Python 2. As for the other things, we've only tested on OSX (x86), Windows, and Ubuntu / CentOS.  I would have to look to see GH has support for ARM-based OSX, if so, we should probably support that given that is what Apple is moving towards. As for i586 et. al, unless we have a runner to validate against in an automated fashion, I don't think we would be able to support stability on those architectures.

hkpeprah, 2023-05-25:
Looks like M1 runners are supported, so this is something we can do for the unit testing: https://github.com/github/roadmap/issues/528